PX4/PX4-Autopilot #16813: CUAV Nora mag calibration crash

coderkalyan, 2021-02-10:
**Describe the bug**
On master, I am unable to calibrate the mag on CUAV Nora. Either during the calibration, or after it is done but before the values are saved, the autopilot reboots itself and I lose the calibration.

**To Reproduce**
Steps to reproduce the behavior:
1. CUAV Nora on master
2. Calibrate mag with QGC
4. See error

**Expected behavior**
The magnetometer should be calibrated correctly.

**Log Files and Screenshots**
No flight log since I wasn't flying :smile: Unfortunately I am not sure how to get more debug information. I tried taking a look at the debug UART shell output but didn't get anything useful. I can try using a debugger or bisecting to find the commit culprit.

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Just a CUAV Nora connected over USB to QGC. Have observed the same thing over telemetry radio as well.

**Additional context**
Please let me know what I can do to help you reproduce this.

LorenzMeier, 2021-02-10:
When you say master, which specific commit was that? There was a recent fix for mag cal on master.

coderkalyan, 2021-02-10:
@LorenzMeier d03d46b8a302152dafabfed9a8073aae13abbcee

coderkalyan, 2021-02-10:
Oh, I see that ef444da68994ca33f338b2d60a49147eb4afaf25 was commited afterwards. Let me try this now.

coderkalyan, 2021-02-10:
@LorenzMeier Still able to reproduce this on master ef444da68994ca33f338b2d60a49147eb4afaf25.

MaEtUgR, 2021-04-21:
From dev call: Could you please run the calibration with full-rate logging running at the same time and share the log.

closed 2021-04-27 by coderkalyan