PX4/PX4-Autopilot #16814: GPS not working in Gazebo SITL when using Iris with additional sensor.

rblilja, 2021-02-10:
Problem:

GPS not working in Gazebo SITL when using Iris with additional sensor, such as e.g. the “iris_irlock” model.

Observations:

I have the same PX4 code on two Ubunto 20.04 machines, running Gazebo 11.2.0.1. The name of the GPS model in the “world” tab in the Gazebo GUI turns up differently:

Machine 1: iris_irlock::gps0
Machine 2: iris_irlock::iris::gps0

On machine 1 the GPS works both for the basic Iris and Iris with additional sensors.

On machine 2 the GPS works only with the basic Iris. GPS model is then named: iris::gps0

I also note that the white wire frame of the GPS model extends largely in the negative Z direction on machine 2 (in case of iris_irlock). Pose in z reads: -1580173.715905

As said, same code on both machines. There might be a difference between dependencies needed by Gazebo (Jijnja2 is at 2.10.1 on both, which I suspected as the iris.sdf is generated). 

Please advise.

Jaeyoung-Lim, 2021-02-11:
Fixed in https://github.com/PX4/PX4-SITL_gazebo/pull/708

TSC21, 2021-02-11:
Ah looks like this was reported before. Good it's now fixed

rblilja, 2021-02-12:
I still have the same issue as originally described. Tested with a fresh clone of the PX4 repository.

I'm getting this error message (with VERBOSE_SIM=1): 

 [Err] [Joint.cc:297] EXCEPTION: Couldn't Find Child Link[gps0::link]

So something is still going south with the GPS..

rblilja, 2021-02-19:
I have tested this further. Cloned a new repo. Upgraded to gazebo 11.3.0.1.

Gazebo now reports the same error (as previously described), and the same joint name structure appears in the model list. But now the GPS works.

Don’t know if this issue shall be considered closed.