PX4/PX4-Autopilot #16816: Gazebo Real Time Factor gradually slow down?

weiyuhe, 2021-02-10:
I was just running the sitl_gazebo simulation by doing: roslaunch px4 mavros_posix_sitl.launch
I also set the export PX4_SIM_SPEED_FACTOR=6 to speed up the simulation real time factor. However, I noticed the Real Time Factor in gazebo kept slowing down; In one hour, it slowed down from 5.9 to 2.2 and show no sign of stopping.

Here is what I got in the terminal:

```
`Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://quadquad:36961/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
[... 274 lines cut ...]
[ INFO] [1612986050.361620888, 8.788000000]: VER: 1.1: Board hardware:      00000001
[ INFO] [1612986050.361769803, 8.788000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1612986050.361925950, 8.792000000]: VER: 1.1: UID:                 4954414c44494e4f
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
[ INFO] [1612986052.722874321, 22.776000000]: WP: mission received
[ WARN] [1612986117.334049578, 332.012000000]: TM : RTT too high for timesync: 12.00 ms.
[ WARN] [1612986129.893172472, 389.516000000]: TM : RTT too high for timesync: 12.00 ms.
[ WARN] [1612986130.493634136, 391.812000000]: TM : RTT too high for timesync: 12.00 ms.
[ WARN] [1612986160.793541995, 514.016000000]: TM : RTT too high for timesync: 16.00 ms.
[ WARN] [1612986162.209792082, 519.212000000]: TM : RTT too high for timesync: 12.00 ms.
[ WARN] [1612986162.586920919, 520.616000000]: TM : RTT too high for timesync: 12.00 ms.
[ WARN] [1612986172.905909480, 554.312000000]: TM : RTT too high for timesync: 12.00 ms.
[ WARN] [1612986174.720695713, 559.816000000]: TM : RTT too high for timesync: 16.00 ms.
[ WARN] [1612986180.008350318, 577.712000000]: TM : RTT too high for timesync: 12.00 ms.
[ WARN] [1612986180.725185403, 580.312000000]: TM : RTT too high for timesync: 12.00 ms.
[ WARN] [1612986188.288141229, 605.116000000]: TM : RTT too high for timesync: 16.00 ms.
[ WARN] [1612986189.336667518, 608.912000000]: TM : RTT too high for timesync: 12.00 ms.
[ WARN] [1612986190.637559233, 613.212000000]: TM : RTT too high for timesync: 12.00 ms.
[ WARN] [1612986192.093774731, 618.016000000]: TM : RTT too high for timesync: 16.00 ms.
[ WARN] [1612986193.557623630, 623.212000000]: TM : RTT too high for timesync: 12.00 ms.
[ WARN] [1612986194.690526466, 627.216000000]: TM : RTT too high for timesync: 16.00 ms.
[ WARN] [1612986223.534815337, 733.812000000]: TM : RTT too high for timesync: 12.00 ms.

`
```

Jaeyoung-Lim, 2021-02-11:
@weiyuhe I am wondering if this is related to a memory leak on the ROS side. Can you try without ROS?

Jaeyoung-Lim, 2021-03-19:
Closing as the original author didn't reply.

@weiyuhe Please reopen if you think this is still an issue

closed 2021-03-19 by Jaeyoung-Lim