PX4/PX4-Autopilot #16820: Automatic px4 launch & kill

drb-avis, 2021-02-11:
Hi all! I am trying to write a code on python that automatically opens and closes px4. I managed to successfully open px4, using subprocess or os library. However, when the script kills the px4 process, the python process is also automatically killed. Does anybody knows a way to kill px4 from python without killing the python process itself?
I am using gazebo 11 and v.1.11.3 of px4, with ROS noetic on Ubuntu 20

Muammer06, 2022-06-05:
Hello 
Are there any update, can you fixed problem. I need a solution for training ML algorithm.

Muammer06, 2022-06-05:
> Herkese selam! Python'da px4'ü otomatik olarak açan ve kapatan bir kod yazmaya çalışıyorum. Alt işlem veya işletim sistemi kitaplığını kullanarak px4'ü başarıyla açmayı başardım. Ancak, komut dosyası px4 sürecini öldürdüğünde, python süreci de otomatik olarak öldürülür. Python sürecinin kendisini öldürmeden px4'ü python'dan öldürmenin bir yolunu bilen var mı? Ubuntu 20'de ROS noetic ile px4'ün gazebo 11 ve v.1.11.3 sürümünü kullanıyorum

Hey are there any solution, I need for training ML algorithm