PX4/PX4-Autopilot #16823: Use double precision for vehicle position in L1 position controller

CarlOlsson, 2021-02-11:
**Describe the bug**
The fact that we convert the current vehicle position to float precision in the fixed wing position controller results in a discretization error which leads to actuator jitter and roll oscillations

The bug will result in different things depending on if the vehicle is flying straight north/east or any other direction. If it is flying straight north/east there will be large step changes is xtrack error and if it is flying in any other direction there will be much more noise in the xtrack_error resulting in rate setpoint jitter which leads to ailron jitter.

Here is a SITL flight using standard VTOL on master
[image]
[image]

**Solution**
Don't convert the current position and waypoints to float in the fixed with position controller

**Additional context**
After making a PR on our fork I wanted to open one here but the fix will require changes also in the rover position controller since it is also using the L1 library, that is out of scope for me now so creating this issue instead

FYI @tstastny This was the bug I mentioned a while back which leads to roll oscillations for a vehicle with very tight L1 tuning

dagar, 2021-02-11:
I want to verify the cost on cortex m4 where we only have single precision floating point hardware, but let's go for it.

LorenzMeier, 2021-02-12:
I remember that double was 8x float and float was 4x int for M4. So higher, but not a total crazy amount.

closed 2021-02-12 by LorenzMeier