PX4/PX4-Autopilot #16839: Unable to start adis16448

Jaeyoung-Lim, 2021-02-12:
**Describe the bug**
The `aid16448 IMU driver is unable to start on a pixhawk4

**To Reproduce**
Steps to reproduce the behavior:
1. `Tools/mavlink_shell.py`
2. Start the driver
````
nsh> adis16448 -S start
WARN  [SPI_I2C] adis16448: no instance started (no device on bus?)
```

**Expected behavior**
IMU driver is started successfully.

**Log Files and Screenshots**
Starting the driver manually fails. 
- I also tried starting the driver in `rc.board_sensors` and disabled all the other sensors but had resulted in the same.
- I have tried different SPI modes / chip selects. The Blue LED lights up on the interface board, which I suspect that it is trying to start the driver
````
nsh> adis16448 -S start
WARN  [SPI_I2C] adis16448: no instance started (no device on bus?)
```

**Drone (please complete the following information):**
- Pixhawk4 with a ADIS16448AMLZ IMU
![PXL_20210212_102341840](https://user-images.githubusercontent.com/5248102/107756945-37a96d00-6d25-11eb-92af-c0436808d4e6.jpg)

**Additional context**

Jaeyoung-Lim, 2021-02-12:
Update: I tried on one of the old PX4 releases that were used in the lab (something close to 1.9.0-beta), and seems to work without any problems, so not a hardware issue

Jaeyoung-Lim, 2021-02-14:
Disabling the CRC check seems to bypass the issue. Looking into what could have gone wrong with the CRC checks

saengphet, 2021-02-16:
Hi @Jaeyoung-Lim , I now use the adis16470 with firmware 1.9.XX (not sure) and it works well. I plan to upgrade firmware to the current stable version. Do you think the same problem will occur or not with adis16470?

Jaeyoung-Lim, 2021-02-16:
@saengphet I don't think the two IMUs are related, so hard to tell. Moreover, this issue has been fixed by https://github.com/PX4/PX4-Autopilot/pull/16855

LorenzMeier, 2021-02-16:
Fixed.

closed 2021-02-16 by LorenzMeier