PX4/PX4-Autopilot #16843: Error with invalid parameters

Jaeyoung-Lim, 2021-02-12:
**Describe the bug**
Ever since https://github.com/PX4/PX4-Autopilot/pull/16796 was merged, I see the following error when running SITL
```
ERROR [parameters] get: param 65535 invalid
ERROR [parameters] get: param 65535 invalid
ERROR [parameters] get: param 65535 invalid
ERROR [parameters] get: param 65535 invalid
```
**To Reproduce**
Steps to reproduce the behavior:
```
make px4_sitl gazebo_rover
```

**Additional context**
@dagar Any ideas what is causing this?

LorenzMeier, 2021-02-12:
Doesn't show for me - I will retry with a clean build:

```

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.149
INFO  [px4] Creating symlink /Users/lorenzmeier/src/PX4-Autopilot/build/px4_sitl_default/etc -> /Users/lorenzmeier/src/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=1060
INFO  [param] selected parameter default file eeprom/parameters_1060
[param] Loaded: eeprom/parameters_1060
 parameter default: MIS_LTRMIN_ALT -1.00 -> 0.01
 parameter default: MIS_TAKEOFF_ALT 2.50 -> 0.01
 parameter default: NAV_ACC_RAD 10.00 -> 2.00
 parameter default: NAV_LOITER_RAD 50.00 -> 2.00
 parameter default: NAV_FW_ALT_RAD 10.00 -> 1000.00
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] Mission #3 loaded, 4 WPs, curr: 3
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [init] Mixer: etc/mixers-sitl/rover_ackermann_sitl.main.mix on /dev/pwm_output0
INFO  [ekf2] starting instance 0, IMU:0 (1310988), MAG:0 (197388)
INFO  [ekf2] starting instance 1, IMU:1 (1310996), MAG:0 (197388)
INFO  [ekf2] starting instance 2, IMU:2 (1311004), MAG:0 (197388)
INFO  [ekf2] starting instance 3, IMU:0 (1310988), MAG:1 (197644)
INFO  [ekf2] starting instance 4, IMU:1 (1310996), MAG:1 (197644)
INFO  [ekf2] starting instance 5, IMU:2 (1311004), MAG:1 (197644)
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2021-02-12/21_05_19.ulg
INFO  [logger] Opened full log file: ./log/2021-02-12/21_05_19.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
```

dagar, 2021-02-12:
It's an uninitialized param handle that was failing silently before. Only the error message is new. I'll track it down.

https://github.com/PX4/PX4-Autopilot/blob/1981519aad6c342ff1e869b1bcf356c131bf46dd/src/lib/parameters/parameters.cpp#L504-L507

closed 2021-02-12 by LorenzMeier