PX4/PX4-Autopilot #16850: Hardware Security of PX4

korkeep, 2021-02-14:
Hi 😀  

I am working on my project of ensuring the hardware security of the Flight Controller. I am planning to use Infinion make Trusted Platform Modules (TPM) to ensure hardware integrity of Pixhawk Cube.  

As far as my understanding is concerned, whenever TPM modules are used in an IOT environment for hardware security, they are used as root of trust.  

Although, I can implement the TPM + Raspberry Pi system, I have no idea how to begin with interfacing a TPM module to Pixhawk. It will also involve changes in PX4 firmware. Considering SPI/I2C interface of TPM module, how can I start building this system?  

Is there any other way by which I can ensure hardware integrity of the Pixhawk Cube? How can I ensure, nobody messes up with my Pixhawk cube and its firmware incase drone gets into the wrong hands?

mrpollo, 2021-02-15:
@korkeep I might be misunderstanding what you are trying to achieve here, but it looks like you are trying to work with the Cube pilots, you need to talk to the manufacturer of the Cube to get support, please try their forums https://discuss.cubepilot.org

closed 2021-02-15 by mrpollo

LorenzMeier, 2021-02-15:
The FMUv5X and FMUv6X autopilot standards have a secure element by NXP - happy to look into that if you are interested. You can reach me on lorenz@px4.io

korkeep, 2021-02-16:
@mrpollo @LorenzMeier  

Thank you for checking :)
As this is my first opensource project, I will do my best👍

LorenzMeier, 2021-02-16:
@korkeep Could you drop me an email on lorenz@px4.io? I'd like to send you further material.

mrpollo, 2021-02-16:
Thanks for reaching out @korkeep you are welcome on this community, let's follow up via email, let's figure out if we can help you.

korkeep, 2021-02-16:
@LorenzMeier @mrpollo
I really appreciate for your guidance.

If I can contribute for this fascinating project,
It will be a great honor for me. 

Are there any things I need to prepare to participate on this project?

Here is my small resume :)
+ PX4-MalSW: https://github.com/korkeep/PX4-MalSW
+ Github: https://github.com/korkeep 

Please take care of your health. 
Thank you sincerely 😀

Aman738, 2023-02-21:
Hello,
I am also working on similar project where hardware security of the Flight Controller is needed.
It will be a great help if someone guide me with it.

Thank you.