PX4/PX4-Autopilot #16852: holybro kakute f7 ,barometer and compass are not work

Simpleman220, 2021-02-14:
board:
holybro kakute f7 v1.5

px4 version:
1.11.3 stable

qgc version:
v4.1.1

I flashed px4 firmware to kakute f7 board followed this tutorial:
https://docs.px4.io/master/en/flight_controller/kakutef7.html

There has no altitude display on qgc:
[image]

And the altitude is unusual:
[image]

Here are some commands and results i tested:

nsh> 
nsh> 
nsh> 
nsh> i2cdetect
Scanning I2C bus: 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- 0d -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- 76 -- -- -- -- -- -- -- -- -- 
nsh> 
nsh> 
nsh> 
nsh> 
nsh> bmp280 status
INFO  [SPI_I2C] Running on I2C Bus 1
bmp280: sample: 2656 events, 2499713us elapsed, 941.16us avg, min 852us max 2181us 167.642us rms
bmp280: measure: 2657 events, 870127us elapsed, 327.48us avg, min 295us max 1546us 93.894us rms
bmp280: comms errors: 0 events
nsh> 
nsh> 
nsh> 
nsh> 
nsh> qmc5883 status
INFO  [SPI_I2C] Running on I2C Bus 1
qmc5883: read: 32965 events, 30246028us elapsed, 917.52us avg, min 848us max 5356us 94.931us rms
qmc5883: com_err: 0 events
interval:  6666 us
INFO  [drivers_magnetometer] /dev/mag device instance: 0
INFO  [drivers_magnetometer] calibration scale: 0.93521 0.98985 1.09912
INFO  [drivers_magnetometer] calibration offset: 0.47361 -0.19991 0.12319
nsh> 
nsh> 
nsh> 
nsh> sensors status
INFO  [sensors] selected gyro: 2162722 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 100, state: OK
INFO  [data_validator]     val:   0.0051, lp:   0.0028 mean dev:   0.0001 RMS:   0.0062 conf:   1.0000
INFO  [data_validator]     val:  -0.0001, lp:   0.0011 mean dev:   0.0005 RMS:   0.0089 conf:   1.0000
INFO  [data_validator]     val:  -0.0044, lp:  -0.0016 mean dev:  -0.0000 RMS:   0.0021 conf:   1.0000

INFO  [sensors] selected accel: 2162722 (0)
INFO  [data_vidator]     no data

INFO  [vehicle_acceleration] selected sensor: 2162722 (0), rate: 988.1 Hz
INFO  [vehicle_acceleration] estimated bias: [0.0000 0.0000 0.0000]
INFO  [sensor_calibration] ACC 2162722 EN: 1, offset: [0.0330 -0.4166 0.3665] scale: [1.0065 0.9949 0.9950]

INFO  [vehicle_angular_velocity] selected sensor: 2162722 (0), rate: 1000.1 Hz
INFO  [vehicle_angular_velocity] estimated bias: [0.0000 0.0000 0.0000]
INFO  [sensor_calibration] GYRO 2162722 EN: 1, offset: [-0.0160 0.0373 0.0009]

INFO  [vehicle_imu] IMU ID: 2162722, accel interval: 999.4 us, gyro interval: 999.4 us
vehicle_imu: accel data gap: 1 events
vehicle_imu: gyro data gap: 1 events
vehicle_imu: accel update interval: 268949 events, 1016.64us avg, min 382us max 2079us 147.690us rms
vehicle_imu: gyro update interval: 273464 events, 1000.00us avg, min 349us max 2079us 161.432us rms
INFO  [sensor_calibration] ACC 2162722 EN: 1, offset: [0.0330 -0.4166 0.3665] scale: [1.0065 0.9949 0.9950]
INFO  [sensor_calibration] GYRO 2162722 EN: 1, offset: [-0.0160 0.0373 0.0009]
nsh> 
nsh> 
nsh> 
nsh> 

The barometer and compass are not work!

Now position,altitude,mission mode cannot work,switch to position mode,error:REJECT POSITION CONTROL

What should I do now?

Thanks,

vipertown, 2021-03-13:
Same issues here.

NuttShell (NSH)
nsh> bmp280 status
INFO  [SPI_I2C] Running on I2C Bus 1
bmp280: sample: 1539 events, 1435750us elapsed, 932.91us avg, min 854us max 1241us 76.170us rms
bmp280: measure: 1540 events, 485198us elapsed, 315.06us avg, min 296us max 852us 34.147us rms
bmp280: comms errors: 0 events
nsh> qmc5883 status
INFO  [SPI_I2C] Not running
nsh> i2cdetect
Scanning I2C bus: 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- 1e -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- 76 -- -- -- -- -- -- -- -- -- 
nsh> sensors status
INFO  [sensors] selected gyro: 2162722 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 100, state: OK
INFO  [data_validator]     val:  -0.0058, lp:  -0.0021 mean dev:   0.0000 RMS:   0.0010 conf:   1.0000
INFO  [data_validator]     val:   0.0064, lp:   0.0065 mean dev:  -0.0000 RMS:   0.0008 conf:   1.0000
INFO  [data_validator]     val:   0.0018, lp:  -0.0000 mean dev:  -0.0000 RMS:   0.0008 conf:   1.0000

INFO  [sensors] selected accel: 2162722 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 100, state: OK
INFO  [data_validator]     val:   1.7818, lp:   1.7685 mean dev:   0.0000 RMS:   0.0173 conf:   1.0000
INFO  [data_validator]     val:   0.2323, lp:   0.2173 mean dev:   0.0001 RMS:   0.0176 conf:   1.0000
INFO  [data_validator]     val:  -9.7465, lp:  -9.7304 mean dev:   0.0006 RMS:   0.0292 conf:   1.0000

INFO  [sensors] selected mag: 73225 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 125, state: OK
INFO  [data_validator]     val:  -0.0637, lp:  -0.0652 mean dev:   0.0000 RMS:   0.0014 conf:   1.0000
INFO  [data_validator]     val:  -0.2554, lp:  -0.2539 mean dev:   0.0000 RMS:   0.0014 conf:   1.0000
INFO  [data_validator]     val:   0.6679, lp:   0.6680 mean dev:  -0.0000 RMS:   0.0014 conf:   1.0000

INFO  [vehicle_air_data] selected barometer: 4158985 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 75, state: OK
INFO  [data_validator]     val: 98210.0703, lp: 98209.7344 mean dev:  -0.3630 RMS:   1.6212 conf:   1.0000
INFO  [data_validator]     val:  35.8868, lp:  35.6034 mean dev:   0.6188 RMS:   0.2497 conf:   1.0000
INFO  [data_validator]     val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   1.0000

INFO  [sensors] Airspeed status:
INFO  [data_validator]     no data

INFO  [vehicle_acceleration] selected sensor: 2162722 (0), rate: 1000.0 Hz
INFO  [vehicle_acceleration] estimated bias: [0.0000 0.0000 0.0000]
INFO  [sensor_calibration] ACC 2162722 EN: 1, offset: [0.0128 -0.2407 0.7193] scale: [1.0031 0.9986 0.9990]

INFO  [vehicle_angular_velocity] selected sensor: 2162722 (0), rate: 1000.0 Hz
INFO  [vehicle_angular_velocity] estimated bias: [0.0000 0.0000 0.0000]
INFO  [sensor_calibration] GYRO 2162722 EN: 1, offset: [-0.0473 -0.0173 0.0024]

INFO  [vehicle_imu] IMU ID: 2162722, accel interval: 998.9 us, gyro interval: 998.9 us
vehicle_imu: accel data gap: 1 events
vehicle_imu: gyro data gap: 1 events
vehicle_imu: accel update interval: 213804 events, 999.92us avg, min 895us max 1106us 3.903us rms
vehicle_imu: gyro update interval: 213809 events, 999.92us avg, min 895us max 1106us 3.904us rms
INFO  [sensor_calibration] ACC 2162722 EN: 1, offset: [0.0128 -0.2407 0.7193] scale: [1.0031 0.9986 0.9990]
INFO  [sensor_calibration] GYRO 2162722 EN: 1, offset: [-0.0473 -0.0173 0.0024]
nsh>

911450448, 2021-12-09:
I haw same problem,no altitude data display,no flight mode can be switched.

911450448, 2021-12-10:
Hi, i find a way to fix this issue.
You need modification your ~/kakutef7/default.cmake, enable ekf2 in your program. This means that you will abandon some function in you FMU.

The kakute_f7 just have 928KB flash to save your user function, may be you shuld enable CONSTRAINED_FLASH to get you program be little.

closed 2025-08-26 by dakejahl