PX4/PX4-Autopilot #16858: EKF2 source RangeFinder does not work as expected

ecmnet, 2021-02-15:
**Describe the bug**
Using odometry as the primary height source in EKF2 and setting PositionControl to 'Terrain following' for Indoor-Use works absolutely as expected. However, using vision data is not as reliable as a LIDAR for height estimation. I case of vision failure, EKF2 falls back to baro estimation. To avoid this, I would like to use LIDAR for height estimation and - as a fallback - optical for XY estimation. This worked prior to 1.11 perfectly, but seems to have changed now in order to support 'Terrain hold'. Is this intended ? 
Another option would be a 'Terrain hold' feature, that uses vision for height estimation and LIDAR for terrain estimation (which usually should not be necessary as vision is independent from the terrain). But in case of Vision failure, a fallback to LIDAR would be possible as long as the vehicle is not moving or - if moving - it would automatically fallback to 'Terrain following'.

ecmnet, 2021-02-15:
Related to https://github.com/PX4/PX4-Autopilot/issues/16526#issue-782676365

bresch, 2021-02-16:
@ecmnet Could you provide a log showing that terrain following problem? We should then be able to see where the missing piece is. What makes you think that "Terrain hold" mode is interfering?

ecmnet, 2021-02-17:
I doubt that any log would help here, as this is a more theoretical consideration. However, I did some more tests. As mentioned above the primary height source is vision. If odometer publishing stops (e.g. while resetting), LPOS-Z estimation does not fall back to Lidar although it is available. I assume, that in that case Z estimation falls back to baro. I try to solve that by sending dummy-odometry data using raw Lidar data (corrected by the last known difference of  Lidar and vision) and current LPOSX/Y. Z estimation now looks quite good but this solution causes XY issues as Flow (which should be a fallback regarding XY) is not the only source of XY and EKF2 assumes absolute XY positioning instead of relative position (as it does if flow would be the only source). The fallback strategy, I would like to achieve is

LPOS.Z:     VisionZ   => LidarZ (corrected by last known offset) => Baro => FailureAction
LPOSXY:    VisionXY => OpticalFlowXY => FailureAction

dagar, 2021-02-17:
I believe EKF2 range aid working with vision height source would come pretty close in feeling some of this.

> This worked prior to 1.11 perfectly,

I'm actually not sure what you're referring to here.

ecmnet, 2021-02-19:
@dagar Just tested the combination range-aid with vision as primary height source (manually moving the vehicle). As soon as I switch off vision data, local-pos-Z is purely relying on acceleration data:

[image]

Not sure, whether this worked earlier as I saw in old logs Lidar being the primary height source.

dagar, 2021-02-19:
Sorry, I should have been explicit, EKF2 range aid doesn't currently work with vision height source.

It's potential a relatively easy piece to fill in if you're able to test and review. https://github.com/PX4/PX4-ECL/blob/164e53bad39e722f4751c754517ef2b5c4488b3a/EKF/control.cpp#L1054-L1074

https://github.com/PX4/PX4-ECL/pull/648

ecmnet, 2021-02-19:
@dagar I‘ll give this a try within the next days

ecmnet, 2021-02-21:
@dagar: Does it make sense to allow this kind of range fallback only if the RangeAidFlag is set?

ecmnet, 2021-02-22:
Doing a `resetHeight()` only, if the previous state was not rng_hgt solves the settling-in issue:

`if (!_control_status.flags.ev_hgt && isRecent(_time_last_ext_vision, 2 * EV_MAX_INTERVAL)) {
			fuse_height = true;
			setControlEVHeight();
			if (!_control_status_prev.flags.rng_hgt) {
			 resetHeight();
			}
			ECL_INFO("Height mode switched to ev");`

Flow scaling also works with this solution. Of course, terrain hold is not guaranteed in the case of fallback - but main goal is to provide a consistent local position during vision resets - for this it should work.

ecmnet, 2021-02-24:
Updated version: https://github.com/ecmnet/ecl/blob/b0b05145e4656e7fd34711405c16615e9681bf4d/EKF/control.cpp#L999

ecmnet, 2021-02-27:
Updated version: 
https://github.com/ecmnet/ecl/blob/237d7bad4b21352d65dcf2e265512fed4024c08f/EKF/control.cpp#L1002

bresch, 2021-03-04:
@ecmnet The fallback from vision to range finder should be done here: https://github.com/ecmnet/ecl/blob/237d7bad4b21352d65dcf2e265512fed4024c08f/EKF/control.cpp#L822 (check how it is done for baro and GPS height)

ecmnet, 2021-03-07:
I will have a look to that

ecmnet, 2021-03-28:
A first check with 1.12-beta1 shows that the provided workaround is not working anymore. I made the following observations:

-  After vision publishing stops, EKF2 reports continuously a reset to current position
- The switch to range finder, proposed here, is not executed
- Terrain estimation is running away in as soon as vision is stopped

Also horizontal fallback behaves differently now:
- After vision publishing stops, it takes about 5 sesonds to switch to relative positioning state
- Flowfusion seems to work as long as EKF2 Is in state absolute Position, but stops as soon EKF2 switches to relative position. Reason for this might be that no height above ground is available anymore

ecmnet, 2021-03-29:
@bresch: Could you have a look into #994

closed 2021-03-31 by bresch