PX4/PX4-Autopilot #16865: External message definitions not recognised by RTPS interface.

rblilja, 2021-02-16:
Description:

Building firmware with "out-of-tree" module and messages fails if aforementioned messages are added to msg/tools/uorb_rtps_message_ids.yaml for inclusion with the RTPS interface.

Steps:

Following this guide, adding both module and messages: https://docs.px4.io/master/en/advanced/out_of_tree_modules.html

It seems that the uORB micro-CDR does not pick up the new messages, when they are added in the msg/tools/uorb_rtps_message_ids.yaml file.

I think the issue originates from src/modules/micrortps_bridge/CMakeList.txt. From what I can see it does only include messages from msg/, not external modules / message definitions.

I'm not at all an expert on cmake, so perhaps better if someone else rectifies this.. :)

Bonus questions:

1) Can parts of uorb_rtps_message_ids.yaml be "out-of-tree"?
2) Were are the message ID ranges (used in uorb_rtps_message_ids.yaml) defined? Build fails if using IDs above 149 for custom messages. 

Thanks!

TSC21, 2021-02-16:
> Steps:
>
> Following this guide, adding both module and messages: https://docs.px4.io/master/en/advanced/out_of_tree_modules.html
>
> It seems that the uORB micro-CDR does not pick up the new messages, [...]

This is really a specific case and it can be extended, but it's not the highest priority. If you currently are interested in using the bridge in your own module, consider adding the msg to /msg and add your module to /src/modules.

> Bonus questions:
> 
> 1. Can parts of uorb_rtps_message_ids.yaml be "out-of-tree"?

Currently no.

> 2. Were are the message ID ranges (used in uorb_rtps_message_ids.yaml) defined? Build fails if using IDs above 149 for custom messages.

https://github.com/PX4/PX4-Autopilot/blob/master/msg/tools/uorb_rtps_classifier.py#L52. But you seem to be using an older version (the limit was 149 before). The id space after 170 (150 on your case) is reserved for alias/multitopics msgs.

rblilja, 2021-02-16:
Understood. Thanks for the swift reply!

closed 2021-02-16 by rblilja