PX4/PX4-Autopilot #16873: non-default SENS_BOARD_ROT broken

bkueng, 2021-02-17:
Setting a non-default board orientation (e.g. `param set SENS_BOARD_ROT 8`) does not work on current master, the attitude and accel data is not rotated (e.g. on Pixhawk 4).

I haven't looked into it any further, but it must have been somewhat recent.

Noticed by @dk7xe

dagar, 2021-02-18:
It seems to be working for me. @dk7xe what are you seeing exactly?

### Master SENS_BOARD_ROT 0
``` Console
INFO  [sensors] selected accel: 3932170 (1)
INFO  [data_validator] validator: best: -1, prev best: -1, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 50, state: OK
INFO  [data_validator]  val:   0.2186, lp:   0.2064 mean dev:  -0.0001 RMS:   0.2373 conf:   1.0000
INFO  [data_validator]  val:  -0.3010, lp:  -0.2970 mean dev:  -0.0000 RMS:   0.2885 conf:   1.0000
INFO  [data_validator]  val:  -9.7073, lp:  -9.7146 mean dev:  -0.0001 RMS:   0.3537 conf:   1.0000
INFO  [data_validator] sensor #1, prio: 50, state: OK
INFO  [data_validator]  val:   0.2329, lp:   0.2152 mean dev:  -0.0000 RMS:   0.2383 conf:   1.0000
INFO  [data_validator]  val:  -0.2627, lp:  -0.2439 mean dev:  -0.0000 RMS:   0.2863 conf:   1.0000
INFO  [data_validator]  val:  -9.8389, lp:  -9.8396 mean dev:  -0.0001 RMS:   0.3541 conf:   1.0000
INFO  [data_validator] sensor #2, prio: 50, state: OK
INFO  [data_validator]  val:   0.1660, lp:   0.1777 mean dev:   0.0001 RMS:   0.2405 conf:   1.0000
INFO  [data_validator]  val:  -0.3044, lp:  -0.3171 mean dev:  -0.0000 RMS:   0.2903 conf:   1.0000
INFO  [data_validator]  val:  -9.7382, lp:  -9.7502 mean dev:  -0.0001 RMS:   0.3564 conf:   1.0000

nsh> listener vehicle_acceleration

TOPIC: vehicle_acceleration
 vehicle_acceleration_s
        timestamp: 192942498  (0.000682 seconds ago)
        timestamp_sample: 192942181  (317 us before timestamp)
        xyz: [0.2247, -0.2423, -9.8092]
```

### Master SENS_BOARD_ROT 24 (pitch 90)
``` Console
INFO  [sensors] selected accel: 4259850 (2)
INFO  [data_validator] validator: best: -1, prev best: -1, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 50, state: OK
INFO  [data_validator]  val:  -9.6934, lp:  -9.7099 mean dev:  -0.0013 RMS:   0.0105 conf:   1.0000
INFO  [data_validator]  val:  -0.2943, lp:  -0.2965 mean dev:  -0.0004 RMS:   0.0095 conf:   1.0000
INFO  [data_validator]  val:  -0.2268, lp:  -0.2060 mean dev:   0.0001 RMS:   0.0094 conf:   1.0000
INFO  [data_validator] sensor #1, prio: 50, state: OK
INFO  [data_validator]  val:  -9.8235, lp:  -9.8395 mean dev:  -0.0003 RMS:   0.0141 conf:   1.0000
INFO  [data_validator]  val:  -0.2536, lp:  -0.2448 mean dev:   0.0001 RMS:   0.0127 conf:   1.0000
INFO  [data_validator]  val:  -0.2101, lp:  -0.2153 mean dev:   0.0004 RMS:   0.0130 conf:   1.0000
INFO  [data_validator] sensor #2, prio: 50, state: OK
INFO  [data_validator]  val:  -9.7462, lp:  -9.7538 mean dev:  -0.0012 RMS:   0.0221 conf:   1.0000
INFO  [data_validator]  val:  -0.3370, lp:  -0.3274 mean dev:   0.0001 RMS:   0.0157 conf:   1.0000
INFO  [data_validator]  val:  -0.1510, lp:  -0.1834 mean dev:   0.0004 RMS:   0.0178 conf:   1.0000

nsh> listener vehicle_acceleration

TOPIC: vehicle_acceleration
 vehicle_acceleration_s
        timestamp: 42105150  (0.000417 seconds ago)
        timestamp_sample: 42105069  (81 us before timestamp)
        xyz: [-9.8073, -0.3309, -0.1848]
```

dk7xe, 2021-02-18:
@dagar please see https://logs.px4.io/plot_app?log=37b3d8d8-117e-4ac9-922d-479a6923f9b3

dk7xe, 2021-02-18:
@dagar to answe your question. Horizon in QGC is upside down. Compasd calibration detects wrong orientation. Remaining sensor calibration fails.

dagar, 2021-02-22:
> @dagar to answe your question. Horizon in QGC is upside down. Compasd calibration detects wrong orientation. Remaining sensor calibration fails.

Thanks @dk7xe, I see the problem now. The board rotation isn't being applied unless the sensor's calibration is found (your log doesn't have an accel cal). Please try https://github.com/PX4/PX4-Autopilot/pull/16919.

dk7xe, 2021-02-23:
@dagar works perfectly! Thank you!
[image]

closed 2021-02-23 by bresch