PX4/PX4-Autopilot #16883: make px4_fmu-v5 upload to CUAV nano error

mengchaoheng, 2021-02-18:
**Describe the bug**
The master branch starts from 318c7e83b33713e2080db27391636cff5ba51a24, upload error on CUAV nano, pwm output is abnormal, and the channel that should output 0 before the safety switch is output 900.

**To Reproduce**
Steps to reproduce the behavior:
1. make distclean && git checkout 318c7e83b33713e2080db27391636cff5ba51a24
2. git submodule update --init --recursive && make px4_fmu-v5 && make px4_fmu-v5 upload
3. in QGC, the vehicle setup is ok
4. After booting up, the initial ringtone is abnormal, in the MAVlink inspector of QGC, the servox_raw of SERVO_OUTPUT_RAW is abnormal

**Expected behavior**
 In the MAVlink inspector of QGC, the servo1_raw and  servo2_raw  of SERVO_OUTPUT_RAW should be 0 before pressing the safety switch.

[image]

**Drone (please complete the following information):**
coaxial helicopter, SYS_AUTOSTART = 15001

mengchaoheng, 2021-02-18:
@Daniel Agar
Do you know the reason？

dagar, 2021-02-18:
The default disarm value is wrong, I'm looking for a system wide fix.