PX4/PX4-Autopilot #16890: Feasibility of Turtle Mode

int-smart, 2021-02-18:
**Describe problem solved by the proposed feature**
Is it possible to get turtle mode functionality (similar to betaflight) in PX4. This functionality is highly desirable in cases where you crash at places like roofs. 

**Describe your preferred solution**
Reverse the motor direction in this mode and enable two of the motors to spin according to direction in which you want to flip.
PX4 does support DSHOT and I believe reversing motor direction should not be a problem.

dagar, 2021-02-18:
Yes with DSHOT it's easy to command reverse.

https://github.com/PX4/PX4-Autopilot/blob/5dc5ebc0a1f5d9cc03b094d2fba24497b1661b72/src/drivers/drv_pwm_output.h#L282-L283

Then next you'd need to implement the basic logic to read from `manual_control_setpoint` and send whatever's appropriate directly to the mixer (actuator_controls_0).

ta-jetson, 2025-02-17:
Hello @dagar @int-smart, Thanks for sharing inputs in this domain. I'm working on something similar. In the current stable version of PX4 1.15.4, is this feature available?

If not available, could I build it? I'm using a FPV ESC on a FC compatible with PX4.

yisrael-benshlomo, 2026-11-09:
We implemented configurable DShot turtle recovery for PX4 and would like to contribute it upstream.

Source, documentation and tests: https://github.com/BENSHLOMO/px4-fpv-extensions

The implementation temporarily reverses the four mapped ESCs, drives the motor(s) selected with the roll/pitch stick, then stops and restores normal direction. Twenty-one `TRTL_*` parameters configure motor mapping, radio input, power, ramp timing and recovery angles through QGroundControl. It includes cancellation handling, switch-OFF qualification between attempts, and Commander/DShot ownership coordination.

Hardware testing covers Dragonfly, our 2.5-inch guarded Quad-X with a MicoAir H743 V2 and AM32 ESCs. The demonstration shows recovery from different orientations, repeated attempts, and subsequent normal arming and flight. The exact AM32 firmware version remains to be recorded.

The BSD-3-Clause source patch targets PX4 v1.16.1, pinned to 94cb2012792b2ae89f0b147cfee53ee31ae550be. The repository includes host tests for controller transitions, RC validity, mapping, cancellation and the arming coordination guard.

We would appreciate maintainer feedback on the preferred upstream architecture, particularly where recovery sequencing and output ownership should live and how they should integrate with Commander. Is there interest in taking this forward as an upstream PR?

Demo:

https://github.com/user-attachments/assets/52dea387-f7dd-4787-bfcc-19a3bb74b075

dakejahl, 2026-11-09:
Very cool! A PR would definitely be appreciated, but I'm not sure how soon the maintainer team will be able to review it.

> The implementation temporarily reverses the four mapped ESCs

Shouldn't you only reverse 2? This way you keep the drone from moving laterally too much? 

My first thought is that this should be a mode (like RTL) so you can flip to that mode via a switch on the RC. Then perhaps the actuator is gated behind throttle stick position, and the vehicle will keep trying to flip (with progressively more throttle?) while the throttle stick is above X position. Then once it's detected as upright the turtle mode goes into an idle state (does nothing).

yisrael-benshlomo, 2026-11-09:
Thanks @dakejahl! I’m preparing an upstream PR.

One clarification: all four ESCs receive the temporary direction command, but only the roll/pitch-selected motors receive throttle—two for a single-axis input, or one for a diagonal. The others receive STOP. This lets the pilot change recovery direction without another reversal sequence.

Currently, recovery uses a dedicated RC switch while disarmed, with throttle held low. Roll/pitch selects direction and can scale power, with configurable ramp and output limits. Once upright, it stops and restores normal motor direction, then requires switch OFF before another attempt.

A dedicated PX4 mode with throttle-controlled effort is worth discussing for upstream. Would you prefer that interface for the initial PR? I can open the current implementation as a draft so we can review the control interface and output ownership together.

dakejahl, 2026-11-09:
I'd say open the PR as you've got it and we can review and go from there

yisrael-benshlomo, 2026-11-11:
@dakejahl The upstream PR is now ready for review: https://github.com/PX4/PX4-Autopilot/pull/29750⁠�
It includes the implementation, documentation and tests. I’d appreciate your feedback on the integration and pilot controls whenever the team has time.