PX4/PX4-Autopilot #16894: SBUS might wrongly get parsed as DSM for the first few frames

bkueng, 2021-02-19:
Since 95dd2f7e8dfca, it's possible that initially SBUS input on the IO gets parsed as DSM, until SBUS locks.

This is the parsed channels I see:
```
got dsm 1250 1747 1200 1752 1200 1769 1089
got dsm 1250 1747 1200 1752 1200 1769 1089
got dsm 1250 1747 1200 1752 1200 1769 1089
got dsm 1250 1747 1200 1723 1454 1912 1089 1974 1043
got dsm 1250 1747 1200 1752 1126 1945 1089 1974 1146 1723
got dsm 1250
got dsm 1250 1747 1043 1752 1200 1945 1089 1974 1442 1723
got dsm 1250 1747 1043 1752 1043 1912 1089 1974 1146
got dsm 1250 1747 1043 1752 1043 1912
got dsm 1743 1747 1043 1752 1043 1769 1089
got dsm 1250 1747 1043 1752 1200
got dsm 1341 1747 1043 1723 1126
INFO  [commander] Manual control regained after 1.7s
INFO  [commander] Kill-switch engaged
got dsm 1341 1747 1043 1723 1126
got dsm 1250 1747 1043 1723 1126 1912 1089 1974 1442
got dsm 1250 1747 1043 1723 1126 1912 1089 1974 1442
WARN  [commander] REJECT POSITION CONTROL
INFO  [commander] Kill-switch disengaged
```

To reproduce:
- v5x, connect SBUS out to SBUS in
- SYS_AUTOSTART = 8001
- toggle `PWM_SBUS_MODE` and look at the data, it happens fairly quickly (~30%)

I don't have a Spektrum receiver that I could test DSM changes.

dagar, 2021-02-19:
We'll have to be careful here. The previous rounds of attempted DSM parser improvements resulted in increased robustness for certain Spectrum setups and complete breakage for others.

Maybe we can simply raise the bar requiring it to successfully decode quite a few more packets continuously before locking.

dagar, 2021-02-19:
We could bump this. https://github.com/PX4/PX4-Autopilot/blob/c863d6a346631d728b3438967f239053f0b2b7ce/src/lib/rc/dsm.cpp#L286-L289

dagar, 2021-04-15:
It's not intended as a fix for this, but https://github.com/PX4/PX4-Autopilot/pull/17401 might improve the situation without reintroducing dependence on the problematic DSM system byte.

bkueng, 2021-04-23:
#17401 didn't improve it.