PX4/PX4-Autopilot #16896: px4flow doesnt work when connected to pixhawk and powered using battery and telemetry

VarunBurde, 2021-02-19:
Hello,

I am having detection issues with the px4flow optical flow sensor when powered via battery.  When pixhawk 4 is powered with battery and connect computer using telemetry radio (px4flow is connected to pixhawk with i2c A )there is no OPTICAL_FLOW_RAD (mavlink messages) published. But when (px4flow sensor is connected to pixhawk 4 via i2c A) and pixhawk is connected to computer via USB cable the OPTICAL_FLOW_RAD (mavlink messages) published and visible through mavlink inspector. The same thing happening with the cubepilot cube orange with firmware v1.11.3. 

Further trying with other firmware versions,  firmware 1.9.1 works with pixhawk 4 with USB as well as over radio telemetry. Could someone please take a look over it and help? 

I will be happy to share logs and further information if required.

Thank you in advance

dagar, 2021-02-21:
With current firmware can you do some additional debugging with the console (over mavlink or real serial console)?

``` Console
nsh> px4flow status

nsh> work_queue status

nsh> listener optical_flow
```

VarunBurde, 2021-02-22:
Hello, 
Thank you for reply.

Here are some snapshot from the mavlink console

When using telemetry setup 

[image]
[image]
[image]

When using usb 

[image]
[image]
[image]

Thank you

VarunBurde, 2021-03-11:
Hello, Any suggestion on how to make it work? @dagar

edwinpha, 2021-03-11:
Hey! I have the very same problem.
I am testing PX4FLOW only with a distance sensor.

Considering that the LOCAL_POSITION_NED is giving a value for X and Y, I believe that the Pixhawk is actually receiving information from PX4FLOW. However, I haven't managed to obtain an accurate movement in the XY-plane.
How is the accuracy of your movements?

Not sure if it is a parameter on the PX4FLOW or on the PIXHAWK to be calibrated.
I am still debugging it, so if anything I will let you know here.

VarunBurde, 2021-03-17:
Hello @edwinpha , Thank you for the suggestion. You are correct, LOCAL_POSITION_NED is getting values for X and Y even there is no OPTICAL_FLOW_RAD data published when connected to px4flow via i2c (x and y are 0 when there is no px4flow connected). Try to connect px4flow over USB and check for the ODOMETRY data for values X and Y. Move the px4flow camera along with pixhawk over high textured uniform space. I think it's not best but doing the job, there is a slight delay but you can see the X and Y values change as you move in one of the axes. Also please try to stick px4 and pixhawk together as they suggest here https://docs.px4.io/master/en/sensor/px4flow.html (Mounting/Orientation section ). Also, have you tried to mount the setup on the aircraft? Were you able to switch to position control mode? Even though there are X and Y data published in LOCAL_POSITION_NED, I cant switch to position control mode.

saengphet, 2021-04-07:
@VarunBurde  @edwinpha  Hello,how about this problem? Can it work for now? I plan to use px4flow too. Thanks

VarunBurde, 2021-04-07:
Hello @saengphet , If you plan to use px4flow with pixhawk 4, it works with firmware v1.9.1. I am unaware of any other autopilot.

julianoes, 2023-04-03:
I'm closing this as it's quite old. However, @iamrajee or anyone else in this thread: if you still have the issue, please open up a new thread and post hardware, software version and a logfile.

closed 2023-04-03 by julianoes