PX4/PX4-Autopilot #16901: Waiting for simulatr to accept connection on TCP port 4560

F-LAB-Systems, 2021-02-20:
```
[100%] Built target px4
Scanning dependencies of target gazebo
SITL ARGS
sitl_bin: /home/anup/PX4_Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /home/anup/PX4_Firmware
build_path: /home/anup/PX4_Firmware/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/home/anup/PX4_Firmware/build/px4_sitl_default/build_gazebo:/home/anup/PX4_Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/anup/PX4_Firmware//Tools/sitl_gazebo/models:/home/anup/PX4_Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /opt/ros/noetic/lib:/home/anup/PX4_Firmware/build/px4_sitl_default/build_gazebo:/home/anup/PX4_Firmware/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::event::Connection; typename boost::detail::sp_member_access<T>::type = gazebo::event::Connection*]: Assertion `px != 0' failed.
SITL COMMAND: "/home/anup/PX4_Firmware/build/px4_sitl_default/bin/px4" "/home/anup/PX4_Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/anup/PX4_Firmware"/test_data
INFO  [px4] Creating symlink /home/anup/PX4_Firmware/ROMFS/px4fmu_common -> /home/anup/PX4_Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
* SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC_PRIME: curr: 0 -> new: 1311244
  CAL_GYRO0_ID: curr: 0 -> new: 2294028
  CAL_GYRO_PRIME: curr: 0 -> new: 2294028
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG_PRIME: curr: 0 -> new: 197388
  COM_DISARM_LAND: curr: 2.0000 -> new: 0.5000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  COM_ARM_EKF_AB: curr: 0.0017 -> new: 0.0050
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.1000 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_SPOOLUP_TIME: curr: 1.0000 -> new: 0.5000
  MPC_TKO_RAMP_T: curr: 3.0000 -> new: 1.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 3 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
* RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
* RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
  GPS_UBX_DYNMODEL: curr: 7 -> new: 6
* SYS_AUTOCONFIG: curr: 1 -> new: 0
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[Err] [Connection.cc:546] Connection[0] Closed during Read
[Err] [ConnectionManager.cc:131] Unable to read from master
[Err] [gazebo_shared.cc:77] Unable to initialize transport.
[Err] [gazebo_client.cc:56] Unable to setup Gazebo

```

facing issue with the connection on TCP port, Please help !!

lelliott20s, 2021-02-22:
Can you launch gazebo standalone with the command gazebo? What commands did you use to get to this point?

Jaeyoung-Lim, 2021-02-22:
> [Err] [Connection.cc:546] Connection[0] Closed during Read
[Err] [ConnectionManager.cc:131] Unable to read from master
[Err] [gazebo_shared.cc:77] Unable to initialize transport.
[Err] [gazebo_client.cc:56] Unable to setup Gazebo

Looks like `gzserver` is not running

CharlChc, 2022-02-15:
Got the same problem, gazebo could be launched standalone. Anybody knows how to solve it? thanks.

Barathwaja, 2022-08-09:
I have the same issue in MacOS Monterey.

Anyone knows how to resolve this?

Thanks !!

ryanjAA, 2022-09-28:
Seeing this today now - I saw it awhile ago and new toolchain fixed it but that isn't doing the trick now

ryanjAA, 2022-09-30:
No. Been using a docker container as a hack for the time  being. 

What  did your install look like? did everything complete without errors? Which OS are you on and Intel based mac?

999zhc999, 2023-08-03:
I got the same error ,please help
[image]

Cristian-wp, 2023-10-12:
Same here, any updates?

julianoes, 2023-10-13:
There is usually another error somewhere that you need to figure out.

For instance in the screenshot [above](https://github.com/PX4/PX4-Autopilot/issues/16901#issuecomment-1663294159) there is an undefined symbol to ignition fuel_tools. So that presumably means the library has changed and a rebuild is required.

ryanjAA, 2023-10-13:
Ya, that always ended up fixing it. Clean, rebuild. A few times I need to reinstall the toolchain too.

Cristian-wp, 2023-10-13:
Hi @julianoes ! How are you :)?
For my case the solution was put in .bashrc: 
`source /usr/share/gazebo-11/setup.sh `
Plus I deleted [px4-sock-0](https://discuss.px4.io/t/sitl-building-issues/9910/12) because was created owned by root (I do not know why).

julianoes, 2023-10-16:
> Plus I deleted [px4-sock-0](https://discuss.px4.io/t/sitl-building-issues/9910/12) because was created owned by root (I do not know why).

Yes, sometimes this one doesn't get cleaned up properly, e.g. when you kill PX4 with -SIGKILL.

Cristian-wp, 2023-10-16:
@julianoes you are always the solution man of all my problems xD

xttzka, 2023-11-14:
> I got the same error ,please help [image]

Have you solved this problems? I have met the same problem , how can I solve it ?

julianoes, 2023-11-14:
There is a "undefined symbol: ignition fuel tools". Make sure these libraries are up to date and do a clean build.

crazy-robotics, 2025-01-07:
I create my custom drone urdf and connect with px4 and my drone is spawn in gazebo simulator. but QGC is not connected

julianoes, 2025-01-07:
@crazy-robotics please don't necrobump this old thread with a different problem and no context to actually help you.

closed 2025-01-07 by julianoes

julianoes, 2025-03-13:
@PuvaneshRam please don't necrobump either, and don't post multiple times. You have been warned.