PX4/PX4-Autopilot #16904: logger: confusing/wrong description of SDLOG_PROFILE parameter

lukegluke, 2021-02-21:
Default value is 1, that is the "Default set" (only bit 0 is set) and this **doesn't** allow estimator replay!
https://github.com/PX4/PX4-Autopilot/blob/414f9f81d93ce48a19092bb73139db7ad333fe8a/src/modules/logger/params.c#L110-L112

dagar, 2021-02-21:
Do you mean checking "Estimator replay (EKF2)" doesn't work without also having "Default set" checked?

lukegluke, 2021-02-21:
I mean that looking at phrase "The default allows for general log analysis and estimator replay", without checking meaning of particular bits, one can decide that default value 1 will be enough for further estimator replay.

dagar, 2021-02-22:
Oh I see what you mean. We used to have estimator replay checked by default as well.

dagar, 2021-02-22:
https://github.com/PX4/PX4-Autopilot/pull/16912

closed 2021-02-22 by bkueng

lukegluke, 2021-03-15:
@dagar, by the way, docs are also not reflect current default:
https://docs.px4.io/master/en/debug/system_wide_replay.html#prerequisites

> All identified topics need to be logged at full rate (see logging). For ekf2 this is already the case with the default set of logged topics.