PX4/PX4-Autopilot #16918: Nora intermittent SD card failures

dagar, 2021-02-22:
There are reports of SD card failures (very likely failing to start in the first place) on the CUAV Nora, which sound quite similar to what we've heard on the Holybro Durandal.

Needs to be reproduced on the bench to see what's actually going on.

Possibly related to power on timing.

https://github.com/PX4/PX4-Autopilot/blob/a8a9832b595321f9073438ae6964fe9ed211d2f4/boards/cuav/nora/src/init.c#L189-L192

sanderux, 2021-03-02:
I have seen no repeats of this issue after cpu issue was fixed

davids5, 2021-03-02:
@sanderux - what is "cpu issue was fixed" referring to?

sanderux, 2021-03-02:
https://github.com/PX4/PX4-Autopilot/pull/16917

sanderux, 2021-03-03:
Reconfirmed, sdcard failure mid mission

sanderux, 2021-03-03:
Here is the log  with data leading up to the sd failure
https://logs.px4.io/plot_app?log=b29b3f33-bb0d-4625-be77-1c1cbfd3b9b3

@RomanBapst @bkueng

davids5, 2021-03-24:
@sanderux  I was told this was card related. With the the SDMMC change is in and the different card is this resolved?

davids5, 2021-04-02:
@sanderux - Please reopen if this is still an issue

closed 2021-04-02 by davids5

sanderux, 2021-04-02:
> @sanderux I was told this was card related. With the the SDMMC change is in and the different card is this resolved?

Looks like it, still need to fly but from bench testing i can confirm i was able to reproduce and solve with faster sd-card.
We use Kingston Industrial Temperature 8GB cards, but they are quite slow.
I'll send you one