PX4/PX4-Autopilot #16928: mocap_pose_estimate does not work on px4 firmware version 1.10.1 with LPE running mavros

ashirsat, 2021-02-23:
**Describe the bug**
Unable to get mocap_pose_estimate plugin to work with optitrack mocap message for position hold.

**To Reproduce**
Steps to reproduce the behavior:
1. I am following the guide for [Using Vision or Motion Capture Systems for Position Estimation](https://docs.px4.io/master/en/ros/external_position_estimation.html) for enabling position set points with optitrack
2. First I launch mavros using the following command `roslaunch mavros px4.launch fcu_url:=///dev/ttyAMA0:921600 gcs_url:=tcp-l://[gcs-ip]:port_id` 
2. Then on a seperate terminal i launch the vrpn_clinet node using the command  `roslaunch vrpn_client_ros sample.launch server:=<server-ip>`
3. Next on a seperate terminal I use topic_tools to relay `/vrpn_client_ros/robot_id/pose` to `/mavros/mocap/pose` topic
4. However I do not get message saying '[lpe] mocap position init ....' and neither anything else happens after this

**Expected behavior**
after realying the optitrack pose message to mocap/pose topic the LPE should do 'mocap position init to ....'

**Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:39903/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
[... 214 lines cut ...]
[ INFO] [1614053400.027897894]: Built-in SIMD instructions: None
[ INFO] [1614053400.028073727]: Built-in MAVLink package version: 2020.11.11
[ INFO] [1614053400.028330862]: Known MAVLink dialects: common ardupilotmega ASLUAV all autoquad icarous matrixpilot paparazzi standard uAvionix ualberta
[ INFO] [1614053400.028600758]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1614053400.035670550]: IMU: High resolution IMU detected!
[ INFO] [1614053400.036255758]: IMU: Attitude quaternion IMU detected!
[ INFO] [1614053400.486167060]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1614053400.494504560]: IMU: High resolution IMU detected!
[ INFO] [1614053400.498253987]: IMU: Attitude quaternion IMU detected!
[ERROR] [1614053400.876154820]: FCU: Mag #1 fail:  STALE!
[ERROR] [1614053400.925605862]: FCU: Mag #2 fail:  STALE!
[ INFO] [1614053401.503866799]: WP: Using MISSION_ITEM_INT
[ INFO] [1614053401.504174091]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1614053401.504625914]: VER: 1.1: Flight software:     010a01ff (e0f016c2b3000000)
[ INFO] [1614053401.504864247]: VER: 1.1: Middleware software: 010a01ff (e0f016c2b3000000)
[ INFO] [1614053401.505040706]: VER: 1.1: OS software:         071d00ff (427238133be2b0ec)
[ INFO] [1614053401.505199924]: VER: 1.1: Board hardware:      00000011
[ INFO] [1614053401.505343518]: VER: 1.1: VID/PID:             26ac:0011
[ INFO] [1614053401.505465445]: VER: 1.1: UID:                 3036511135323931
[ WARN] [1614053401.505726383]: CMD: Unexpected command 520, result 0
[ INFO] [1614053410.486802525]: HP: requesting home position
[ INFO] [1614053415.491673773]: WP: mission received
[ INFO] [1614053420.486753146]: HP: requesting home position
**
Any suggestion to resolve this will be appreciated

ashirsat, 2021-02-23:
@TSC21 any suggestions on this?

TSC21, 2021-02-23:
@ashirsat is there any reason you are using LPE? We are not actively maintaining LPE anymore, so I would suggest you use EKF2 instead and send the mocap data through the `odom` or `vision_pose_estimate` plugins.

ashirsat, 2021-02-25:
@TSC21 It was an easier estimator to work with and for debugging. No other reason. Also I tried  `vision_pose_estimate` I still have an offset of 90 deg when I work with optitrack system. So I do not know if the transformations are done correctly with the vision_pose topic on the mavros side. I have followed the guide mentioned in the initial bug report.

TSC21, 2021-02-25:
> @TSC21 It was an easier estimator to work with and for debugging. No other reason. Also I tried `vision_pose_estimate` I still have an offset of 90 deg when I work with optitrack system with respect to the NED frame. The heading direction reported by px4 and verified by QGC is east when I pointing North( +x axis of the optitrack system). So I do not know if the transformations are done correctly with the vision_pose topic on the mavros side. I have followed the guide mentioned in the initial bug report.

That's probably because the optitrack system is providing the orientation in a left-handed reference or not in a different frame of reference than the expected. Try to swap the rotations in https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/vision_pose_estimate.cpp#L100-L101

ashirsat, 2021-02-25:
@TSC21 Is this code a part of the binary installation with `ros-melodic-mavros-estras`?

TSC21, 2021-02-25:
> @TSC21 Is this code a part of the binary installation with `ros-melodic-mavros-extras`?

What code?

ashirsat, 2021-02-25:
@TSC21  The lines that you linked in vision-pose-estimate.ccp file?

ashirsat, 2021-02-25:
> > @TSC21 Is this code a part of the binary installation with `ros-melodic-mavros-extras`?
> 
> What code?

The lines that you linked in the previous response?

ashirsat, 2021-02-25:
> That's probably because the optitrack system is providing the orientation in a left-handed reference or not in a different frame of reference than the expected. Try to swap the rotations in [...]

Also I don't think it is a left hand co-ordinate system. If I follow the guide I think I get +x forward +y left of pixhawk and +z upwards.