PX4/PX4-Autopilot #16930: Unstable flight - Shaking from side to side

gizmond1982, 2021-02-23:
Hi all,

I have just completed my first S500 drone build. I am using 40A ESC, T-motor 2216, Pixhawk 2.4.8, m8n gps, and 5200mah / 60C lipo.

My drone can takeoff and land but during the flight the drone is very jumpy and unstable shaking from side to side (left, right, front, back) even almost no winds.

I was wondering what could be the reason...

Here are the logs - 
https://logs.px4.io/plot_app?log=8dfa4940-16f4-4934-b91b-281926d05ce9

notoriou5, 2021-02-28:
@gizmond1982 
On the first look it seems like a PID tuning issue. Especially the roll attitude can not be followed at all:
[image]
Ideally, the red line should exactly follow the green like. I guess the MC_PITCH_P param is set to high. Did you do a PID tuning or did you just fly with some preset parameters. If you are new to the hobby, maybe read up on how to PID tune (e.g. on the [PX4 docs homepage](https://docs.px4.io/master/en/config_mc/pid_tuning_guide_multicopter.html)).

Hope this helps a bit :)

gizmond1982, 2021-02-28:
Thank you @notoriou5 .
I was able to improve it with in-flight PID tuning. Not perfect but flyable.
https://review.px4.io/plot_app?log=8bc698c9-a17c-4952-b3cc-64195cbbf725

bresch, 2021-03-04:
That's already much better @gizmond1982 !
[image]

As this isn't a problem with the firmware I'll close this issue, but feel free to ask for more help on the forum (http://discuss.px4.io) and/or slack if needed.

closed 2021-03-04 by bresch