PX4/PX4-Autopilot #16934: Copter flips when manually moving it after safety switch is pressed

marvin-wtt, 2021-02-23:
**Describe the bug**
When trying to take off after moving the copter on the ground, the copter flips. It seems like the drone wants the correct its position go get to the previous land position.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on drone
2. Go into position mode
3. Press safety switch
4. Rotate drone in yaw angle
5. Turn off kill switch and arm
6. Take off

**Expected behavior**
Drone takes of in position mode and follows manual control input.
Corrections should be reset when arming drone

**Log Files and Screenshots**
- [Log File](https://review.px4.io/plot_app?log=c05e1bc2-6a02-445b-8e79-6e255f769231)
- I can provide additional camera footage if needed.

**Drone (please complete the following information):**
- Tarot Iron man 680 TL68C01
- Pixhawk is installed between frame bottom and battery
- GPS with magnetometer is installed on top

bresch, 2021-02-23:
It looks like the drone tries to reach the initial yaw angle, not the position.

[image]

marvin-wtt, 2021-02-23:
Yes. You are right. However is this behavior intentional?

bresch, 2024-04-03:
was fixed by https://github.com/PX4/PX4-Autopilot/pull/16051 since v1.12, closing

closed 2024-04-03 by bresch