PX4/PX4-Autopilot #16942: UAVCAN v0 Node Issue Tracker and Discussion

mrpollo, 2021-02-24:
The following Issue is an effort to continue tracking the discussions and progress of the UAVCAN v0 Node.

The CAN Node for v0 support has been in the codebase for a while, without much use until now, the developer team has put an extra effort to bring it back to full support, and is currently testing it with hardware from different vendors Including CUAV, Holybro, and NXP, all of which will be releasing hardware in the near future with UAVCAN v0 support using the PX4 CAN Node

The project is being led by @davids5 and @LorenzMeier, with the help from @dagar

Progress Tracking

* [x] flash based parameter storage
* [x] safety button UAVCAN node support
* [x] RGB LED UAVCAN node support
* [x] WS2812B?
* [ ] check mag orientation
* [x] bootloader
* [x] watchdog
* [x] App desc / HW ID support
* [x] Refactor fw server
* [x] CUAV Hardware Working
* [x] Holybro Hardware Working
* [ ] NXP Hardware Working

Relevant PRs

The PX4 cannode bootloader resurrection
https://github.com/PX4/PX4-Autopilot/pull/16554

**UAVCAN Node Register/Parameter Server**
https://github.com/PX4/PX4-Autopilot/pull/16649
https://github.com/PX4/PX4-Autopilot/pull/16595

**DNM S32K can bootloader**

https://github.com/PX4/PX4-Autopilot/pull/16879

igalloway, 2021-02-24:
@mrpollo - what do you mean by Hardware Support - "working"
The UCANS32K146 software is working well. It was just the bootloader(s) left to do. The CAN bootloader has a PR now (as you have noticed.

mrpollo, 2021-02-24:
hey @igalloway  It was a reference to the bootloader PR from @davids5, we got it merged :)

igalloway, 2021-02-24:
Super! Thanks

mzahana, 2021-03-02:
@mrpollo I faced this issue (https://github.com/PX4/PX4-Autopilot/issues/17004)
And I thought it's related. I hope to get some help on that.

Thanks!

mrpollo, 2021-03-02:
Thanks for the report @mzahana, I replied to your issue, it seems unrelated.

fnax8902, 2021-03-02:
Which release/commit introduced RGB-LED support?

mrpollo, 2021-03-02:
@fnax8902 LED support was added to the CUAV Neo v3 PRO through various commits, but the gist of it is on this commit https://github.com/PX4/PX4-Autopilot/pull/16554/commits/a96fafffd8b3ee8116d46cb488cf7ce0429fa1be

davids5, 2021-03-02:
That is just the original bootloader one and it was replaced, with https://github.com/PX4/PX4-Autopilot/pull/16619

fnax8902, 2021-03-03:
RGB LED is working with the Here 3 GPS using current master, thank you!

mzahana, 2021-03-04:
@fnax8902 Which autopilot hardware did you use?
Would it work on the Orange Cube?

fnax8902, 2021-03-04:
I do have a holybro pix32 and according to the ulg-file I'm using "PX4_FMU_V3 (V2)".

julianoes, 2023-04-03:
I would assume this can now be closed?

dk7xe, 2023-04-03:
> I would assume this can now be closed?

What about the uORB messages?

julianoes, 2023-04-03:
@dk7xe what exactly is to do there, is there a use case that requires it? I feel like it would have been implemented if it was important, right?