PX4/PX4-Autopilot #16944: Always getting failsafe enabled: no offboard warning

96moustafa, 2021-02-24:
I using ROS2 foxy to publish position_setpoints in OFFBOARD mode as in the offboard_control example. The multicopter moves to the desired points as expected. However, I always get the warning from the commander:

` WARN  [commander] Failsafe enabled: no RC and no offboard`

I have no RC connected, but I don't understand why warns me about "no offboard". I know that I should publish at a certain rate and COM_OF_LOSS_T parameter specifies the time-out after which the offboard lost action will be triggered (currently it is 0.5s). 

While running the offboard_control example, I checked the publishing frequency using `ros2 topic hz /PositionSetpointTriplet_PubSubTopic` and the average rate is 100hz (the worst rate is 76hz) which should be more than enough. 

I also checked the uORB rate in pxh shell using `uorb top -1` and it is as follows:
```
TOPIC NAME                         INST #SUB RATE #Q SIZE
offboard_control_mode                 0    2  101  1   24
position_setpoint_triplet             0   11  104  1  320
```

I tried to increase the value of COM_OF_LOSS_T to see the behavior and found that when it is at 1s I get the warning way less frequently.

mengchaoheng, 2022-10-23:
The current flight failsafe strategy is very unreasonable, I can't take off many planes at the same time, because I can't ensure multiple data links in the simulation @Jaeyoung-Lim

Jaeyoung-Lim, 2022-10-23:
@mengchaoheng why cant you establish datalinks for multiple datalinks in simulation?

If you want, you can turn the failsafe off. However, I dont understand why such a failsafe is unreasonable

mengchaoheng, 2022-10-23:
How to turn off failsafe? NAV_RCL_ACT doesn't seem to be allowed to be set to 0. I want to run ros node in off board mode to control multiple uavs, like https://docs.px4.io/main/en/simulation/multi_vehicle_simulation_gazebo.html. But failsafe seems to be holding me back. @Jaeyoung-Lim

Jaeyoung-Lim, 2022-10-23:
> How to turn off failsafe? NAV_RCL_ACT doesn't seem to be allowed to be set to 0.

@mengchaoheng You can disable the datalink failsafe as explained here: https://docs.px4.io/main/en/config/safety.html#data-link-loss-failsafe

> I want to run ros node in off board mode to control multiple uavs, like https://docs.px4.io/main/en/simulation/multi_vehicle_simulation_gazebo.html. But failsafe seems to be holding me back. @Jaeyoung-Lim

If you have multiple vehicles connected to QGC, you should not have this failsafe trigger in the first place

mengchaoheng, 2022-10-24:
Described in https://github.com/PX4/PX4-Autopilot/issues/20469, maybe the qgc have bug. @Jaeyoung-Lim

mengchaoheng, 2022-10-24:
> why cant you establish datalinks for multiple datalinks in simulation?
> 
> If you want, you can turn the failsafe off. However, I dont understand why such a failsafe is unreasonable

It seems that qgc can only provide links for one uav, even if it recognizes multiple aircraft at the same time, so it can only make one uav take off at a time.

Jaeyoung-Lim, 2022-10-24:
> It seems that qgc can only provide links for one uav, even if it recognizes multiple aircraft at the same time, so it can only make one uav take off at a time.

Although I don't think this topic is not relevant for the original issue, I am not sure if I agree. You can send vehicle commands one by one for each aircraft, but QGC can communicate with multiple vehicles at the same time.