PX4/PX4-Autopilot #16945: Unable to compile to the code

ibrahimqazi, 2021-02-24:
Hello dear developers,
I am unable to compile to code from source, no matter I compile for which board I always get the error; Can someone please guide what is the root cause of this error?
`make px4_fmu-v3_default
-- PX4 version: v1.11.0-rc3-1684-gcec31fd685
-- PX4 config file: /media/ibrrahim/New Volume/PX4-Autopilot/boards/px4/fmu-v3/default.cmake
-- PX4 config: px4_fmu-v3_default
-- PX4 platform: nuttx
Makefile.host:23: /media/ibrrahim/New: No such file or directory
Makefile.host:23: Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/../tools/Config.mk: No such file or directory
make[1]: *** No rule to make target 'Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/../tools/Config.mk'.  Stop.
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: [: /media/ibrrahim/New: unexpected operator
/usr/bin/env: ‘bash\r’: No such file or directory
/usr/bin/env: ‘bash\r’: No such file or directory
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: [: /media/ibrrahim/New: unexpected operator
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/bin/sh: 1: /media/ibrrahim/New\ Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/incdir: not found
/usr/bin/env: ‘bash\r’: No such file or directory
make[2]: *** [clean_context] Error 127
make[1]: *** [.clean_context] Error 2
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 9.3.1
-- The C compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working C compiler: /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
CMake Error at platforms/nuttx/cmake/px4_impl_os.cmake:128 (message):
  Could not determine chip architecture from NuttX config.  You may have to
  add it.
Call Stack (most recent call first):
  CMakeLists.txt:304 (px4_os_determine_build_chip)

-- Configuring incomplete, errors occurred!
See also "/media/ibrrahim/New Volume/PX4-Autopilot/build/px4_fmu-v3_default/CMakeFiles/CMakeOutput.log".
See also "/media/ibrrahim/New Volume/PX4-Autopilot/build/px4_fmu-v3_default/CMakeFiles/CMakeError.log".
Error: /media/ibrrahim/New Volume/PX4-Autopilot/build/px4_fmu-v3_default is not a directory
Makefile:217: recipe for target 'px4_fmu-v3_default' failed
make: *** [px4_fmu-v3_default] Error 1
`

bresch, 2021-02-24:
```
Makefile.host:23: /media/ibrrahim/New: No such file or directory
Makefile.host:23: Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/../tools/Config.mk: No such file or directory
```
there is a space between `New` and `Volume`

ibrahimqazi, 2021-02-25:
> ```
> Makefile.host:23: /media/ibrrahim/New: No such file or directory
> Makefile.host:23: Volume/PX4-Autopilot/build/px4_fmu-v3_default/NuttX/nuttx/tools/../tools/Config.mk: No such file or directory
> ```
> 
> there is a space between `New` and `Volume`

Thanks it was a strange error.

closed 2021-02-25 by bresch