PX4/PX4-Autopilot #16950: 'make px4_fmu-v4' fails - compilers not found

jmalbert97, 2021-02-24:
I am attempting to simply build the firmware for a NuttX / PixHawk board, following the guide here: https://docs.px4.io/master/en/dev_setup/building_px4.html. I have the firmware cloned, the toolchains installed, but the build fails. I should note, when I build for JMavSim, it builds just fine. When I build for an actual NuttX target (I am using PixHawk 4) the build fails. I have tried adding environmental variables (verified with echo) to point to the paths for each compiler… but output is exactly the same, not even showing the path i entered. Here is my output:

```
joe@joe-ThinkPad-T440p:~/PX4-Autopilot$ sudo make px4_fmu-v4
-- PX4 version: v1.11.0-rc3-829-g3276916df9
-- PX4 config file: /home/joe/PX4-Autopilot/boards/px4/fmu-v4/default.cmake
-- PX4 config: px4_fmu-v4_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is unknown
-- The C compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: arm-none-eabi-gcc
CMake Error at CMakeLists.txt:202 (project):
  The CMAKE_CXX_COMPILER:

    arm-none-eabi-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:202 (project):
  The CMAKE_C_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:202 (project):
  The CMAKE_ASM_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "/home/joe/PX4-Autopilot/build/px4_fmu-v4_default/CMakeFiles/CMakeOutput.log".
See also "/home/joe/PX4-Autopilot/build/px4_fmu-v4_default/CMakeFiles/CMakeError.log".
Error: /home/joe/PX4-Autopilot/build/px4_fmu-v4_default is not a directory
Makefile:224: recipe for target 'px4_fmu-v4' failed
make: *** [px4_fmu-v4] Error 1
```

My GCC version:

```
joe@joe-ThinkPad-T440p:~/PX4-Autopilot$ gcc --version
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

‘find -iname arm-none-eabi*’ output, showing these compilers do in fact exist (altough, seems like a weird place for them)
```
.......
./opt/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-elfedit
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc-ar
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-ar
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-strings
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-c++
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-c++filt
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc-9.3.1
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-readelf
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-addr2line
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc-ranlib
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-ranlib
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-cpp
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-objcopy
./opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
.......
```
Any guidance here would be appreciated. I am compiling on Ubuntu 18.04.

dvireshlumenier, 2021-03-02:
I would recommend running the setup script for your particular os. For ubuntu it is ./Tools/setup/ubuntu.sh it is possible that your compiler part is not linked properly and that script should fix it. Make sure to log out and log back in once you run that script.

afwilkin, 2021-03-28:
I am experiencing this issue as well. The ubuntu.sh setup did not work for me. 

Is there any update on this?

afwilkin, 2021-03-28:
I found the issue. On the last version of PX4-autopilot I had to run with `sudo` because the program needed permission to create a cache file `Failed to create temporary file for /home/atlas/.ccache/5/d/ec4da3686341d3d3968582736b61b9-11825.o.tmp.stdout: Permission denied`

So, I simply ran with `sudo`. In this new version this seems to screw things up. 

I am working on a fix, will update

afwilkin, 2021-03-28:
Yeah, I am unsure why permissions on my folder `/home/atlas/.ccache/5/d/` were set for only root use, but after changing the permissions there I was able to get a successful build

julianoes, 2021-03-29:
> I am experiencing this issue as well. The ubuntu.sh setup did not work for me.

Please paste the log of what did not work.

jnomikos, 2023-07-06:
I ran into this issue today, and found out it was because I ran ubuntu.sh in Tools/setup with sudo, instead of as a normal user.

superBaik, 2023-09-13:
Hi there, 
Even though I performed this process with "sudo',  I m faced with the exactly same issus now. 
can I know there is any possible way to pass this error ? 

thx in advance

mcsauder, 2023-09-13:
> Hi there, Even though I performed this process with "sudo', I m faced with the exactly same issus now. can I know there is any possible way to pass this error ?
> 
> thx in advance

reboot, update via `sudo apt -y update` reboot, and it might fix your issue. if not, then reboot again.  If still not fixed, try

`sudo apt -y update &&
sudo apt -y upgrade &&
sudo apt -y autoremove &&
sudo apt -y clean &&
sudo apt -y autoclean
`

If not fixed, post your new status.

wuyanze0226, 2024-10-28:
hello ,i  ran into this issue today,anyone fixed it? How to fix it?Thank you.

julianoes, 2024-10-28:
This issue is old. The steps how to install the toolchain are documented here:
https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html#simulation-and-nuttx-pixhawk-targets

If it doesn't work for you @wuyanze0226 please create a new thread and give all the information about your system and the steps you tried, so that there is a chance we can help you. Necrobumping an old thread doesn't help here.