PX4/PX4-Autopilot #16968: Potential division-by-zero in airspeed.cpp

wuruoyu, 2021-02-26:
**Describe the bug**
In function `calc_IAS_corrected()` in `airspeed.cpp`, under `case AIRSPEED_COMPENSATION_TUBE_PRESSURE_LOSS`, at line 143, division-by-zero exception can happen if `eps` equals to `-1`. And to me it is possible given the code above it, which is the following. Although I am not sure about the scale of each variable, it is easy to see that `eps` can be -1. 
```
if (fabsf(denominator) > 1e-32f) {
  const float viscosity = (18.205f + 0.0484f * (temperature_celsius - 20.0f)) * 1e-6f;
  // 4.79 * 1e-7 -> mass flow through sensor
  // 59.5 -> dp sensor constant where linear and quadratic contribution to dp vs flow is equal
  eps = -64.0f * tube_len * viscosity * 4.79f * 1e-7f * (sqrtf(1.0f + 8.0f * dp / 59.3319f) - 1.0f) / denominator;
}
```
I use a solver to solve in which condition `eps` can be `-1`, following is what the solver gives back to me:
[image]
It seems to me that these values of variables are reasonable. 

**Expected behavior**
A guarding check before the division would be preferred here.

bresch, 2021-02-26:
The check is already there:
https://github.com/PX4/PX4-Autopilot/blob/84455ac9e35f872db88b67dce5b08be7923b9b00/src/lib/airspeed/airspeed.cpp#L130-L133

wuruoyu, 2021-02-26:
Oh, I missed that. Thank you~

closed 2021-02-26 by wuruoyu