PX4/PX4-Autopilot #16985: Potential division-by-zero in mc_pos_control/PositionControl.cpp

wuruoyu, 2021-02-27:
**Describe the bug**
There is a potential division-by-zero exception in the following code. 

https://github.com/PX4/PX4-Autopilot/blob/9d47f7ecda2153cc3fde3a2a63096723b3467651/src/modules/mc_pos_control/PositionControl/PositionControl.cpp#L184

The `_hover_thrust` can be zero according to the comment below

https://github.com/PX4/PX4-Autopilot/blob/9d47f7ecda2153cc3fde3a2a63096723b3467651/src/modules/mc_pos_control/PositionControl/PositionControl.hpp#L203

**Expected behavior**
A guarding check before the division would be preferred here.

dayjaby, 2021-02-27:
That's not possible due to `_hover_thr = math::constrain(hover_thrust, 0.1f, 0.9f);`

wuruoyu, 2021-02-27:
I think you are referring to this line:
https://github.com/PX4/PX4-Autopilot/blob/9d47f7ecda2153cc3fde3a2a63096723b3467651/src/modules/mc_hover_thrust_estimator/zero_order_hover_thrust_ekf.hpp#L81

But the `setHoverThrust()` the code in `PositionControl.cpp` is using is the one below, which does not constrain the range. And from the comment of the below function, we can see that zero is within its valid range (i.e. [0, 1]). It would be better to impose the same constrain check as the code above. 

https://github.com/PX4/PX4-Autopilot/blob/9d47f7ecda2153cc3fde3a2a63096723b3467651/src/modules/mc_pos_control/PositionControl/PositionControl.hpp#L116-L120

Thank you for your reply.

closed 2021-03-05 by bresch