PX4/PX4-Autopilot #17004: UaVCAN: Here 3 GPS requires re-plugging to get recognized

mzahana, 2021-03-02:
**Describe the bug**
I am trying to use Here 3 GPS with Cube Orange running PX4 1.11.3. I followed the Cube Pilot [docs](https://docs.cubepilot.org/user-guides/here-3/here-3-manual) to setup both the GPS receiver and PX4, the GPS is not recognized  unless I unplug and then replug the GPS CAN cable while the Cube orange is ON. Also, whether I do software or hardware reboot to the Cube orange, it does not get recognized.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Here 3 to either CAN1 or CAN2 ports and do the UAV configuration as mentioned in the Here 3 manual 
2. Drone switched on 
2. Connect to QGC
3. No GPS is recognized (no satellites)
4. While drone is ON, unplug the Here 3, and replug it to either CAN1/2
5. GPS is recognized and number of satellites is available

**Expected behavior**
* connect Here 3 to CAN port
* Power on the autopilot and do the UAVCAN configuration, and then reboot?
* Here 3 should be recognized

JacobCrabill, 2021-03-02:
I've actually had this issue with other autopilot boards as well - I'm guessing it's a problem with the Here3, but I'm not certain.

mrpollo, 2021-03-02:
I found the following on the cube pilot forums, which hopefully clears up the situation

https://discuss.cubepilot.org/t/here3-gps-needs-to-be-unplugged-and-re-pluggged-everytime-i-power-on-quadcopter-with-herelink-px4-orange-cube/5547/3

amikhalev, 2021-03-02:
Have you tried updating firmware on the Here3 and u-blox module?
https://docs.cubepilot.org/user-guides/here-3/here-3-manual#1-here3-firmware-update

mzahana, 2021-03-03:
@amikhalev Thank you for the suggestion! 
Updating the Here3 firmware solved the hardware re-plugging issue. So, at least I can just reboot PX4 from QGC and Here3 gets recognized!

The remaining issue is to be able the Here3 directly after power up the system, without having to send a reboot command from QGC.

mzahana, 2021-03-03:
Unfortunately, the problem is still there. I though it was solved, but it actually only worked right after defining a PX4 Airframe and before completing the remaining vehicle configuration (sensor & radio calibration ... etc). After completing the vehicle configuration, the Here3 issue appeared again.

amikhalev, 2021-03-03:
If the issue is related to boot timing that would make sense, as more things are started once the airframe is configured.

PX4 doesn't have a `BRD_BOOT_DELAY` equivalent but it might be worth trying to add a `sleep 1` to one of the boot RC files. I also remember reading something about moving the UAVCAN initialization later/earlier in the boot process, but I can't seem to find it now.

mzahana, 2021-03-04:
I got the Here3 GPS working when I provided external 5v to the TELEM2 port, while I was using that port for onboard computer communication. Now the GPS works right a way after powering up the orange cube as expected. So, I am guessing it’s a power issue or the way The Cube handles power to the I/Os.

ThomasRigi, 2021-03-15:
I've had the same or at least a similar issue: https://github.com/PX4/PX4-Autopilot/issues/16113 

At the time it was working on master, so no effort was going in on fixing it on v1.11. I did some tests on my side and found that with a different (slower) SD card it worked, but abandoned further tests after having killed two Pixhawk 4s.

Can you try on master and/or v1.12beta?

matthewgan, 2021-10-05:
Had the same issue using v1.12.2 release.
with Cube Orange and Here 3 GPS

eldewany, 2021-12-11:
any updates on this ? i'm on 1.12.3 and after updating everything (Here3, u-blox chip) nothing has changed, not even replugging works. The GPS gets no fix at all.

dagar, 2021-12-12:
> any updates on this ? i'm on 1.12.3 and after updating everything (Here3, u-blox chip) nothing has changed, not even replugging works. The GPS gets no fix at all.

@eldewany are you seeing any data from the Here3 at all? I don't have a Here3, but I've been regularly using other UAVCAN GPS units with no issue.

mzahana, 2021-12-13:
@dagar I am interested to know what UAVCAN GPS modules you are using. Are they compatible with PX4?

mzahana, 2021-12-13:
@dagar And, are they RTK modules?

dagar, 2021-12-13:
> @dagar I am interested to know what UAVCAN GPS modules you are using. Are they compatible with PX4?

My point was more that there isn't anything fundamentally broken with PX4 UAVCAN GPS handling, there are a growing number of units to choose from (F9P option https://arkelectron.com/product/ark-rtk-gps/). 

I don't have a Here3 GPS, but if anyone is up for a remote debug session I'd be happy to try and get to the root cause here. There's no reason it shouldn't be fully compatible, it sounds like there must be something a little odd happening at bootup with this unit.

eldewany, 2021-12-13:
@dagar Here's the full case, i have an ark flow installed on CAN1 and works fine, I tried installing the Here3 on CAN 2 , but it wasn't recognized at all so I attached it to the secondary CAN port on the arkflow (Thus both are linked on CAN1), i tested 2 units of the HERE3 that worked natively on firmware 1.12.1 with Here 3 fw 1.5, when I upgraded px4 to 1.12.2 or 1.12.3 it stopped working (0 satellites) until I updated the Here3 FW to 1.8 (latest), one of the units worked and the other didn't.

eldewany, 2021-12-13:
I also tried the replug method, but still nothing. It's also worth mentioning that the Here3 is detected in both units after updating but one of them never received a signal although it was working fine before updating the px4 firmware that was later followed by the Here3 fw update.

dagar, 2021-12-13:
@eldewany can you confirm if you're getting data from the unit when it reports 0 satellites? For example `listener sensor_gps` from the PX4 console vs what QGC shows in the bar?

PX4 UAVCAN is responsible for dynamic node id allocation, but after that it's little more than a bridge between UAVCAN => uORB. 

If the units are "working", but not getting a fix I'm not sure what PX4 could be doing wrong.

thomasteisberg, 2022-05-20:
Is this issue solved? With an updated Here 3 (running firmware 1.8), everything seems to "just work" (so far, at least).

As far as I can tell, there are no firmware release notes for the Here 3, so I don't know if it's plausible that this issue was fixed on the Here 3 side sometime between 1.6 and 1.8.

(For anyone possibly Googling this, you do seem to still have to follow the instructions for setting a CAN ID using Ardupilot and Mission Planner here: https://docs.cubepilot.org/user-guides/here-3/here-3-manual and I also had to use the trick explained here to get Mission Planner to detect the Here 3 device: https://discuss.cubepilot.org/t/slcan-mode-not-working/7483/7)

julianoes, 2023-04-03:
I just tested this using a CubeOrange with latest main/future v1.14.

I could not reproduce the issue. I powered the CubeOrange with either USB or without USB with a power module connected.

Both times, the GPS was recognized just fine, also after a `reboot`.

I tried Here3 versions 1.5, 1.7 and 1.8.

I took the file from here:
https://github.com/CubePilot/Here2/releases/download/v1.8/Here2_com.hex.here_3.0-crc.bin

And flashed it using the DroneCAN GUI and a USB to CAN adapter.

closed 2023-04-03 by julianoes