PX4/PX4-Autopilot #17006: SITL jMAVsim oscillates in pitch

MaEtUgR, 2021-03-02:
**Describe the bug**
When using master I observed small but steady and fast oscillations already in hover flight in simulation from a certain point in time on. I never had the time to follow where exactly this comes from and the sim was still usable for the normal tests I do although the oscillations are annoying and could affect observations in tests. When using master today the same oscillations became a lot worse and rendered it unusable so I started investigating.

**To Reproduce**
Steps to reproduce the behavior:
1. Start `make px4_sitl jmavsim`
2. Wait until the simulated GPS is accepted
3. Command a takeoff
4. See pitch oscillations when looking at either the level hovering vehicle's attitude in the visualization or the attitude indicator of the ground station.

**Expected behavior**
Nice smooth normal flight without oscillations.

**Investigation**
- Before https://github.com/PX4/PX4-Autopilot/pull/16188 it's flying exactly as expected. Smooth motion, no oscillations.
- After the D-term gyro filter cutoff was reduced it has the small annoying oscillation.
- After https://github.com/PX4/PX4-Autopilot/pull/15820 specifically https://github.com/PX4/PX4-Autopilot/commit/6482120d9a9b144831c81e7d2206b561fe8ad1d3 SITL jMAVsim is unusable.

Now since I didn't do any real-world test flights the question is:
- Are there general regressions that are not exclusive to my tests?
- Are the defaults generally less suited now?
- Are the defaults just less suited for SITL jMAVsim now and retuning it (which I'll happily do) is the only thing needed.

FYI @dagar @bresch @bkueng

dagar, 2021-03-02:
I'll follow up tomorrow, but in general the current defaults are intended to be relatively safe by default (look at vehicle_angular_acceleration in a typical user setup).

MaEtUgR, 2021-03-03:
- It's purely because of `IMU_DGYRO_CUTOFF` being too low. If I change that back during SITL flight it's instantly fixed. General regressions ruled out.
- Interestingly it's pitch only where it is that sensitive. We're using the same default D-term gain of 0.003. Using the new default of `IMU_DGYRO_CUTOFF` 10Hz I have to increase the roll D-term gain to 0.008 to get a similarly bad roll flight behavior like pitch with the default gain. Assuming that the jMAVsim drone is symmetric (to be confirmed) that's quite suspicious.

mehmetyldz87, 2021-03-09:
Hi @MaEtUgR 
I've done a few test flights on this issue. With v1.11.3, the value of  **_IMU_DGYRO_CUTOFF_**  has been updated to **0**. 

### TEST Flight 1
**IMU_DGYRO_CUTOFF**  = 0
https://logs.px4.io/plot_app?log=a13b6a53-1006-4d2a-bdbb-ff03040768ee 

### TEST Flight 2
**IMU_DGYRO_CUTOFF**  = 10  [Default]
 https://logs.px4.io/plot_app?log=f69ff845-f6fa-4704-a35f-57b922441f70 

### TEST Flight 3
**IMU_DGYRO_CUTOFF**  = 15  
https://logs.px4.io/plot_app?log=5c5ce078-834f-41fd-b03a-a4c9f6ae8c0f 

The higher this value of IMU_DGYRO_CUTOFF* , the better the flight performance.

MaEtUgR, 2021-04-06:
https://github.com/PX4/PX4-Autopilot/pull/17287 was a not so nice hotfix.
https://github.com/PX4/PX4-Autopilot/pull/17290 amkes sense in general but does not solve this issue.
I really wonder why the issue is pitch only. I also tested with increased `IMU_GYRO_CUTOFF` (50-60Hz). The two filters might have some strange interaction on the pitch noise model in jMAVSim. I'll check further.

closed 2021-05-17 by julianoes