PX4/PX4-Autopilot #17007: Battery failsafe kicks in too early

sanderux, 2021-03-02:
Log: https://logs.px4.io/plot_app?log=2c9c93ec-1b0a-49f7-beeb-ce98f10700d9

Battery alert was issued at 37% capacity remaining, BAT_LOW_THR was set to 20%

Other tests show that battery failsafe RTL actually kicks in around 32% while 15% was set

battery_status.warning was never above 0 indicating the battery failsafe was not issued by the battery sensor code.

@dagar @julianoes @RomanBapst

sanderux, 2021-03-02:
this branch is last weeks master but also incorporates this pr: https://github.com/PX4/PX4-Autopilot/pull/16933

sanderux, 2021-03-02:
Looks like it is related to this: https://github.com/PX4/PX4-Autopilot/blob/c2151cb4fafc8fdbbf136d6da4989185b0763217/src/modules/commander/Commander.cpp#L3861

sanderux, 2021-03-02:
and here is some strange math going on: https://github.com/PX4/PX4-Autopilot/blob/ed7a5314d85c76bd922bad559e74f0ceaabbb840/src/modules/navigator/rtl.cpp#L229

somehow the "time to get home" is compared to the battery percentage which says very little about flight time

bresch, 2021-03-03:
[RTL_FLT_TIME](https://dev.px4.io/master/en/advanced/parameter_reference.html#RTL_FLT_TIME):
> This is used to determine when the vehicle should be switched to RTL due to low battery. Note, particularly for multirotors this should reflect flight time at cruise speed, not while stationary

sanderux, 2021-03-03:
Ok, i;ll test.
Should we address the potential division by zero here?
float rtl_flight_time_ratio = time_to_home_s / (60 * _param_rtl_flt_time.get());

bresch, 2021-03-03:
> Should we address the potential division by zero here?

Yes, absolutely

sanderux, 2021-03-03:
ok, so indeed the RTL_FLT_TIME needs to be set to the maximum flight time if the vehicle in total, not the "time needed for RTL"
I'll do a pr fixing the div/0 and provide some better commenting