PX4/PX4-Autopilot #17013: make nxp_fmuk66-e_rtps fails

dk7xe, 2021-03-03:
**Describe the bug**
make nxp_fmuk66-e_rtps fails with error 

**To Reproduce**
Steps to reproduce the behavior:

Prerequisit is that PX4 build environment is set up
Issue the following commands to build
1. cd PX4-Autopilot
2. git pull
3. git submodule update --recursive
4. make nxp_fmuk66-e_rtps
5. build process terminates with error. - see below

**Expected behavior**
make builds for the nxp_fmuk66-e_rtps target

**Log Files and Screenshots**
```
[27/1395] git submodule src/modules/micrortps_bridge/micro-CDR
[1374/1395] Building CXX object src/modules/micrortps_brid...rtps_bridge__micrortps_client.dir/microRTPS_client.cpp.obj
FAILED: src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_client.cpp.obj
/usr/bin/ccache /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++  -DCONFIG_ARCH_BOARD_NXP_FMUK66_E -DCONSTRAINED_MEMORY -DMODULE_NAME=\"micrortps_client\" -DPX4_MAIN=micrortps_client_app_main -D_SYS_CDEFS_H_ -D_SYS_REENT_H_ -D__CUSTOM_FILE_IO__ -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I../../boards/nxp/fmuk66-e/src -I../../platforms/nuttx/src/px4/common/include -I. -Isrc/lib -I../../platforms/nuttx/src/px4/nxp/k66/include -I../../platforms/common -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include -I../../src/modules/micrortps_bridge/micrortps_client -Isrc/modules/micrortps_bridge/micrortps_client -I../../src/modules/micrortps_bridge/micro-CDR/include -Isrc/modules/micrortps_bridge/micro-CDR/include -Isrc/modules/micrortps_bridge/micro-CDR/include/microcdr -isystem ../../platforms/nuttx/NuttX/include/cxx -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Os -DNDEBUG   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-truncation -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -nostdinc++ -fno-sized-deallocation -Wframe-larger-than=4096 -std=gnu++14 -MD -MT src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_client.cpp.obj -MF src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_client.cpp.obj.d -o src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_client.cpp.obj -c src/modules/micrortps_bridge/micrortps_client/microRTPS_client.cpp
arm-none-eabi-g++: error: src/modules/micrortps_bridge/micrortps_client/microRTPS_client.cpp: No such file or directory
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
[1375/1395] Building CXX object src/modules/micrortps_brid...s_bridge__micrortps_client.dir/microRTPS_transport.cpp.obj
FAILED: src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_transport.cpp.obj
/usr/bin/ccache /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++  -DCONFIG_ARCH_BOARD_NXP_FMUK66_E -DCONSTRAINED_MEMORY -DMODULE_NAME=\"micrortps_client\" -DPX4_MAIN=micrortps_client_app_main -D_SYS_CDEFS_H_ -D_SYS_REENT_H_ -D__CUSTOM_FILE_IO__ -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I../../boards/nxp/fmuk66-e/src -I../../platforms/nuttx/src/px4/common/include -I. -Isrc/lib -I../../platforms/nuttx/src/px4/nxp/k66/include -I../../platforms/common -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include -I../../src/modules/micrortps_bridge/micrortps_client -Isrc/modules/micrortps_bridge/micrortps_client -I../../src/modules/micrortps_bridge/micro-CDR/include -Isrc/modules/micrortps_bridge/micro-CDR/include -Isrc/modules/micrortps_bridge/micro-CDR/include/microcdr -isystem ../../platforms/nuttx/NuttX/include/cxx -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Os -DNDEBUG   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-truncation -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -nostdinc++ -fno-sized-deallocation -Wframe-larger-than=4096 -std=gnu++14 -MD -MT src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_transport.cpp.obj -MF src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_transport.cpp.obj.d -o src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_transport.cpp.obj -c src/modules/micrortps_bridge/micrortps_client/microRTPS_transport.cpp
arm-none-eabi-g++: error: src/modules/micrortps_bridge/micrortps_client/microRTPS_transport.cpp: No such file or directory
arm-none-eabi-g++: fatal error: no input files
compilation terminated.
[1377/1395] Building CXX object src/modules/micrortps_brid...bridge__micrortps_client.dir/microRTPS_client_main.cpp.obj
FAILED: src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_client_main.cpp.obj
/usr/bin/ccache /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++  -DCONFIG_ARCH_BOARD_NXP_FMUK66_E -DCONSTRAINED_MEMORY -DMODULE_NAME=\"micrortps_client\" -DPX4_MAIN=micrortps_client_app_main -D_SYS_CDEFS_H_ -D_SYS_REENT_H_ -D__CUSTOM_FILE_IO__ -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I../../boards/nxp/fmuk66-e/src -I../../platforms/nuttx/src/px4/common/include -I. -Isrc/lib -I../../platforms/nuttx/src/px4/nxp/k66/include -I../../platforms/common -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include -I../../src/modules/micrortps_bridge/micrortps_client -Isrc/modules/micrortps_bridge/micrortps_client -I../../src/modules/micrortps_bridge/micro-CDR/include -Isrc/modules/micrortps_bridge/micro-CDR/include -Isrc/modules/micrortps_bridge/micro-CDR/include/microcdr -isystem ../../platforms/nuttx/NuttX/include/cxx -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Os -DNDEBUG   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-t [... cut ...]
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[1383/1395] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/modules__mavlink_unity.cpp.obj
ninja: build stopped: subcommand failed.
make: *** [Makefile:219: nxp_fmuk66-e_rtps] Error 1
```

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
https://github.com/PX4/PX4-Autopilot/pull/17005

davids5, 2021-03-03:
I am building it locally, CI is building it 

[image]

Assuming you do you have any assets in the the source tree, that are not checked in, you can do a `make distclean` and rebuild 
Jut use caution, as it will remove all the cruft, so be careful of hidden files from your ide etc - they may be deleted.

davids5, 2021-03-03:
Issue was on OP machine.

closed 2021-03-03 by davids5

dk7xe, 2021-03-03:
indeed the issue was an error 60.

..the problem sits 60 cm in front of the computer.
Build environment got broken due to an update of the WLS.