PX4/PX4-Autopilot #17021: system - wide replay error

mengchaoheng, 2021-03-04:
**Describe the bug**
replay failed。

**To Reproduce**
Steps to reproduce the behavior:
1. make px4_sitl jmavsim, amd Record the log
2. export replay_mode=ekf2
3. export replay=<abs_path_to_log.ulg>
4. make px4_sitl none
4. See error: ERROR [ekf2] start failed (-1)

**Expected behavior**

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=3b5926dd-2926-4365-b5c3-dc3be6018016

the replay logfille is : https://logs.px4.io/plot_app?log=b504edc0-1fde-44c7-b2df-ede4f487bd6c

mengchaoheng, 2021-03-05:
[image]

NAmmann, 2022-04-01:
Hi @mengchaoheng,

It looks like you are doing the replay of a multi ekf flight log.

The documentation says: The recording and [EKF2 replay](https://docs.px4.io/master/en/debug/system_wide_replay.html#ekf2-replay) of flight logs with multiple EKF instances is not supported. To enable recording for EKF replay you must set the parameters to enable a [single EKF instance](https://docs.px4.io/master/en/advanced_config/tuning_the_ecl_ekf.html#running-a-single-ekf-instance).

So you have to set the parameters as described [here](https://docs.px4.io/master/en/advanced_config/tuning_the_ecl_ekf.html#running-a-single-ekf-instance).

julianoes, 2022-08-02:
No follow up, closing.

closed 2022-08-02 by julianoes