PX4/PX4-Autopilot #17027: git submodule update --init --recursive

Fllying, 2021-03-05:
**Describe the bug**
When download the code from git in linux, I met the following question. So I install a Git.exe on Windows, the question still occur:
git submodule update --init --recursive

**Question**
 git submodule update --init --recursive
Cloning into 'C:/Users/Administrator/Firmware/boards/atlflight/cmake_hexagon'...
Cloning into 'C:/Users/Administrator/Firmware/mavlink/include/mavlink/v2.0'...
fatal: unable to access 'https://github.com/mavlink/c_library_v2.git/': OpenSSL SSL_read: Connection was reset, errno 10054
fatal: clone of 'https://github.com/mavlink/c_library_v2.git' into submodule path 'C:/Users/Administrator/Firmware/mavlink/include/mavlink/v2.0' failed
Failed to clone 'mavlink/include/mavlink/v2.0'. Retry scheduled
Cloning into 'C:/Users/Administrator/Firmware/platforms/nuttx/NuttX/apps'...
fatal: unable to access 'https://github.com/PX4/NuttX-apps.git/': Failed to connect to github.com port 443: Timed out
fatal: clone of 'https://github.com/PX4/NuttX-apps.git' into submodule path 'C:/Users/Administrator/Firmware/platforms/nuttx/NuttX/apps' failed
Failed to clone 'platforms/nuttx/NuttX/apps'. Retry scheduled
Cloning into 'C:/Users/Administrator/Firmware/platforms/nuttx/NuttX/nuttx'...
fatal: unable to access 'https://github.com/PX4/NuttX.git/': Failed to connect to github.com port 443: Timed out
fatal: clone of 'https://github.com/PX4/NuttX.git' into submodule path 'C:/Users/Administrator/Firmware/platforms/nuttx/NuttX/nuttx' failed
Failed to clone 'platforms/nuttx/NuttX/nuttx'. Retry scheduled
Cloning into 'C:/Users/Administrator/Firmware/platforms/qurt/dspal'...
fatal: unable to access 'https://github.com/ATLFlight/dspal.git/': Failed to connect to github.com port 443: Timed out
fatal: clone of 'https://github.com/ATLFlight/dspal.git' into submodule path 'C:/Users/Administrator/Firmware/platforms/qurt/dspal' failed
Failed to clone 'platforms/qurt/dspal'. Retry scheduled
Cloning into 'C:/Users/Administrator/Firmware/src/drivers/gps/devices'...
fatal: unable to access 'https://github.com/PX4/PX4-GPSDrivers.git/': Failed to connect to github.com port 443: Timed out
fatal: clone of 'https://github.com/PX4/PX4-GPSDrivers.git' into submodule path 'C:/Users/Administrator/Firmware/src/drivers/gps/devices' failed
Failed to clone 'src/drivers/gps/devices'. Retry scheduled
Cloning into 'C:/Users/Administrator/Firmware/src/drivers/uavcan/libuavcan'...
Cloning into 'C:/Users/Administrator/Firmware/src/drivers/uavcan_v1/libcanard'...
fatal: unable to access 'https://github.com/UAVCAN/libcanard.git/': OpenSSL SSL_read: Connection was reset, errno 10054
fatal: clone of 'https://github.com/UAVCAN/libcanard.git' into submodule path 'C:/Users/Administrator/Firmware/src/drivers/uavcan_v1/libcanard' failed
Failed to clone 'src/drivers/uavcan_v1/libcanard'. Retry scheduled
Cloning into 'C:/Users/Administrator/Firmware/src/drivers/uavcan_v1/public_regulated_data_types'...
fatal: unable to access 'https://github.com/UAVCAN/public_regulated_data_types.git/': Failed to connect to github.com port 443: Timed out
fatal: clone of 'https://github.com/UAVCAN/public_regulated_data_types.git' into submodule path 'C:/Users/Administrator/Firmware/src/drivers/uavcan_v1/public_regulated_data_types' failed
Failed to clone 'src/drivers/uavcan_v1/public_regulated_data_types'. Retry scheduled
Cloning into 'C:/Users/Administrator/Firmware/src/drivers/uavcannode_gps_demo/libcanard'...
fatal: unable to access 'https://github.com/UAVCAN/libcanard/': Failed to connect to github.com port 443: Timed out
fatal: clone of 'https://github.com/UAVCAN/libcanard' into submodule path 'C:/Users/Administrator/Firmware/src/drivers/uavcannode_gps_demo/libcanard' failed
Failed to clone 'src/drivers/uavcannode_gps_demo/libcanard'. Retry scheduled
Cloning into 'C:/Users/Administrator/Firmware/src/drivers/uavcannode_gps_demo/public_regulated_data_types'...
fatal: unable to access 'https://github.com/UAVCAN/public_regulated_data_types/': OpenSSL SSL_read: Connection was reset, errno 10054
fatal: clone of 'https://github.com/UAVCAN/public_regulated_data_types' into submodule path 'C:/Users/Administrator/Firmware/src/drivers/uavcannode_gps_demo/public_regulated_data_types' failed
Failed to clone 'src/drivers/uavcannode_gps_demo/public_regulated_data_types'. Retry scheduled
Cloning into 'C:/Users/Administrator/Firmware/src/examples/gyro_fft/CMSIS_5'...
fetch-pack: unexpected disconnect while reading sideband packet

Can anyone fix it? thanks a lot.

dagar, 2021-03-05:
It looks like a connection timeout. Can you try again?

Fllying, 2021-03-08:
There is no good way, even I have chosen a different network proxy. Like cmake_hexagon, nuttx etc.

bys1123, 2021-03-08:
Please use VPN in china.

bys1123, 2021-03-08:
Welcome to join chinese PX4 developer WeChat Group 微信群，我的微信号是253331754

Fllying, 2021-03-09:
> Welcome to join chinese PX4 developer WeChat Group 微信群，我的微信号是253331754

Thanks a lot and we were friend,  ha

closed 2021-03-09 by Fllying

Taolinyinjiu, 2024-05-23:
2024/5/23