PX4/PX4-Autopilot #17030: Helicopter Airframe can't be used "Failed to communicate with IO, abort"

max11gen, 2021-03-05:
**Describe the bug**
I tried to configure the Airframe `Helicopter --> Blade 130X` on my Pixhawk 4. I added the line `set MIXER_AUX pass` to the airframe so that I could use my RC, but the error occurs without that line as well. I used the current master to flash the pixhawk. When the Pixhawk boots it issues the beeping noise that, according to [this](https://docs.px4.io/v1.9.0/en/getting_started/tunes.html), means that PX4IO failed. After wards you can see an error message that says "Failed to communicate with IO, abort". Afterwards, the output pins don't react to any RC input.
Rebooting with the safety switch held down doesn't change anything and yields the same result.
Another related problem that I found is that you can't load the blader130.main.mix file into any airframe. When you try to do `mixer load /dev/pwm_output0 /etc/mixers/blade130.main.mix` in any other airframe, you get the following error messages:

> WARN  [px4io] fmu sent: "0
> S: 220 13054 10000 0 -8000 8000
> M: 1
> S: 0 2 10000 10000 0 ;"
> ERROR [px4io] mixer send error -1
> ERROR [mixer] failed to load mixers from /etc/mixers/blade130.main.mix
> ERROR [mixer] failed to load mixer

**To Reproduce**
Steps to reproduce the behavior:
1. Open QGC
2. Set Airframe Blade 130X
3. Click "Apply and restart"
4. See error

**Expected behavior**
I expected to not see a "failed to communicate with IO, abort" message and also that my servos will move when I provide flight control input.

**Log Files and Screenshots**
[image]

**Drone (please complete the following information):**
- I just got a Pixhawk with a couple of servos connected. No actual drone.

max11gen, 2021-03-12:
FYI: This bug has already been reported in October in #15984

dagar, 2021-03-12:
> FYI: This bug has already been reported in October in #15984

Thanks for reporting @max11gen. We're in the process of a fairly large px4io refactor to fully address these problems. https://github.com/PX4/PX4-Autopilot/pull/16444

bkueng, 2022-10-25:
IO refactoring done, please reopen if still an issue

closed 2022-10-25 by bkueng