PX4/PX4-Autopilot #17039: Attitude estimation

poudyalbot, 2021-03-06:
Why is the ekf based attitude estimation not used in px4 as it was during the initial phase of px4.
Thank you

dagar, 2021-03-06:
> Why is the ekf based attitude estimation not used in px4 as it was during the initial phase of px4.
> Thank you

The current default (and only officially supported) estimator is ekf2 which is an attitude & position estimator.

https://github.com/PX4/PX4-ECL/blob/4bad2a272cfbcfd2ef2c77bb4af788f5b569aaaa/EKF/common.h#L378-L387

Are you looking for something in particular?

poudyalbot, 2021-03-07:
> f5b569aaaa/EKF/common.h#L378-L387

I was looking for ekf based attitude estimation solution in px4 as it was
in starting of the px4 project.
Thank you

closed 2021-03-09 by bresch