PX4/PX4-Autopilot #17046: Accelerometer Loss During Takeoff Causes Fast Crash

obicons, 2021-03-07:
**Describe the bug**
I experienced a total loss of accelerometers during takeoff. Before I could react, the vehicle crashes.

**To Reproduce**
1. Drone turned on
2. Takeoff in stabilized offboard mode
3. Trigger accelerometer failure

**Expected behavior**
Navigation should proceed using dead reckoning. I should be warned there is no healthy accelerometer so that I can intervene. 

**Log Files and Screenshots**

https://drive.google.com/file/d/14NiXIGAtC6-si2unROaLbm27GCTnaX2f/view?usp=sharing

**Drone (please complete the following information):**
SITL, Gazebo Iris.

**Additional context**
N/A

dagar, 2021-03-08:
https://logs.px4.io/plot_app?log=9a9d4900-746e-448f-a5a7-7b98a9e2f23f

Can you verify the closest upstream PX4 version (the commit isn't on GitHub)?

dagar, 2021-03-08:
Based on the parameters I'd guess it's pretty old. If you try newer PX4 there are multiple IMUs in SITL (like most flight controllers we support) and if the current primary fails it will switch to the next healthy option.

obicons, 2021-03-08:
Thanks so much for you fast response. I tested on master -- I can confirm the issue is not present.

closed 2021-03-08 by obicons