PX4/PX4-Autopilot #17057: Gyro Glitch Causes SITL to hang

obicons, 2021-03-08:
**Describe the bug**
When Gyroscope is lost (e.g. Gyro stops publishing while read errors are accumulated), SITL hangs. "ERROR [simulator] poll timeout 0, 22" is written to stdout repeatedly. If the drone is flying, no failsafes are triggered and the vehicle does not disarm. 

**To Reproduce**
Steps to reproduce the behavior:
1. Drone runs in SITL
2. Gyro stops publishing data
3. SITL hangs

**Expected behavior**
SITL to proceed executing without gyro readings.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=012ba851-07ed-48a4-afe0-c93a302101c9

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
SITL Gazebo

LorenzMeier, 2021-03-08:
How do you know it is the gyro? I've looked at the log and it seems to all stop around the same time. What am I missing?

obicons, 2021-03-08:
I have a custom gyro driver. When that driver stops publishing readings, all of SITL locks -- hence why the log stops at the same time. I think that SITL should continue without the gyros publishing.

LorenzMeier, 2021-03-08:
Then you need to disable lockstep, which is designed to keep SITL in sync when the simulator is falling behind: https://github.com/PX4/PX4-Autopilot/blob/fddcb73802d0b5565796e42b042cd41818282f1c/boards/px4/sitl/default.cmake#L117

closed 2021-03-08 by LorenzMeier