PX4/PX4-Autopilot #17077: make px4_sitl none_iris fails

brunopinto900, 2021-03-10:
Hello,

I am trying to use PX4 SITL with AirSim+ AirSim ROS Wrapper. This last one works just fine, i can visualize an image from a quadrotor in AirSim in Rviz.

I followed the instructions on this link https://microsoft.github.io/AirSim/px4_sitl/ to build the PX4 SITL with AirSim as the simulator. I am running both AirSim ROS Wrapper and PX4 STIL on Ubuntu20.04 in wsl2. I am running AirSim on Windows 10. 

 When i run the command "make px4_sitl none_iris" on the directory ~/PX4/Firmware i have the following output:
 
 [0/1] cd /home/bruno/PX4/Firmware/build/px4_sitl_default/tmp && /home/bruno/PX4/Firmware/Tools/sitl_run...._default/bin/px4 none none iris /home/bruno/PX4/Firmware /home/bruno/PX4/Firmware/build/px4_sitl_defaul
SITL ARGS
sitl_bin: /home/bruno/PX4/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: none
model: iris
src_path: /home/bruno/PX4/Firmware
build_path: /home/bruno/PX4/Firmware/build/px4_sitl_default
SITL COMMAND: "/home/bruno/PX4/Firmware/build/px4_sitl_default/bin/px4" "/home/bruno/PX4/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/bruno/PX4/Firmware"/test_data
INFO  [px4] Creating symlink /home/bruno/PX4/Firmware/ROMFS/px4fmu_common -> /home/bruno/PX4/Firmware/build/px4_sitl_default/tmp/rootfs/etc
ERROR [px4_daemon] error binding socket
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
Error: no autostart file found (etc/init.d-posix/ERROR [px4] PX4 daemon not running yet_*)
ERROR [px4] Startup script returned with return value: 256
pxh>

On AirSim/path_to/settings.json i have "LocalHostIp": "127.0.0.1" (both on windows and ubuntu)

Thanks for your time.

Jaeyoung-Lim, 2021-03-14:
@brunopinto900 I am unable to reproduce your problem. I suspect that this is coming from your setup

I can confirm that this works for Ubuntu 20.04 and Ubuntu 18.04

```
$ make px4_sitl none_iris
[1/2] cd /home/jaeyoung/src/PX4-Autopi...c/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/jaeyoung/src/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: none
model: iris
world: none
src_path: /home/jaeyoung/src/PX4-Autopilot
build_path: /home/jaeyoung/src/PX4-Autopilot/build/px4_sitl_default
SITL COMMAND: "/home/jaeyoung/src/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/jaeyoung/src/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/jaeyoung/src/PX4-Autopilot"/test_data
INFO  [px4] Creating symlink /home/jaeyoung/src/PX4-Autopilot/build/px4_sitl_default/etc -> /home/jaeyoung/src/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* SYS_AUTOCONFIG: curr: 1 -> new: 0
  IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
```

brunopinto900, 2021-03-15:
Hello,

Thanks for your support. I will do the setup from the start. However i am using WSL1, does this work for WSL?

Jaeyoung-Lim, 2021-03-15:
@brunopinto900 Unfortunately I don't have access to windows, therefore I am not able to test with WSL

brunopinto900, 2021-03-15:
> @brunopinto900 Unfortunately I don't have access to windows, therefore I am not able to test with WSL

No problem. I went with this approach https://docs.px4.io/master/en/dev_setup/dev_env_windows_cygwin.html . Its working.

closed 2021-03-16 by Jaeyoung-Lim