PX4/PX4-Autopilot #17081: Appending a mission item to a current mission

fnax8902, 2021-03-10:
Let's suppose I'm flying a mission and while already being in the air I want to append another waypoint/mission item to the mission. If I do this right know via QGC it tells me that the mission has to be paused, so it can be uploaded again. Why isn't it possible to just continue with the mission and just add the item at the end?
In my opinion one should be able to do this.

To archive this I guess a new MAVSDK message, which is only used for appending to an existing mission, would have to be created and then proccesed by PX4.

Right now I'm always uploading a new mission to add something to the existing one, which isn't optimal.

dagar, 2021-03-29:
We can make this possible PX4 side, but it will need to be coordinated with QGC and/or MAVSDK to be useful.

dagar, 2021-03-29:
Oh I remember now, one missing piece is that the GCS doesn't know if its in sync with the vehicle or not, so the safe thing to do is resend the full mission.

We should be able to finally fix that in the near future. https://github.com/mavlink/mavlink/pull/1611

fas2037, 2021-03-29:
That sounds great, I would be really interested in that. I posted the question in the general slack channel and @julianoes mentioned that https://mavlink.io/en/messages/common.html#MISSION_WRITE_PARTIAL_LIST could be used to archieve this.