PX4/PX4-Autopilot #17082: Crazyflie 2.1 EKF2 with flowboard and position mode

TheLegendaryJedi, 2021-03-10:
Like we talked about during [dev call](https://discuss.px4.io/t/px4-dev-call-march-10-2021/21282),

I'm using EK2 with `EKF2_AID_MASK` using only optical flow. I can fly it in `manual` and `stabilized` with no problem. But right now I want to make use of the flow board. I wanted to takeoff and hold position. I can't change to position mode, it just don't give me any feedback.

I' using these configurations (already the dafault on `master`):
```
param set-default SYS_MC_EST_GROUP 2
param set-default SYS_HAS_MAG 0
param set-default EKF2_AID_MASK 2
param set-default EKF2_MAG_TYPE 5
```
I can see that in QGC the distance sensor is working fine. But right now I don't know what to test or debug next to achieve takeoff and hold.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Flash the master branch into Crazyflie 2.1
2. Open QGC
3. Try to change to position mode

Thanks

jinger26, 2021-03-24:
I'll work on getting @bresch some hardware

TheLegendaryJedi, 2021-03-24:
@bresch I tried to change the `EK2_MAG_TYPE` to `MC custom` and I still can't change to position mode

bresch, 2021-03-25:
@TheLegendaryJedi Yes, because EKF2 waits for mag data to complete the initialization even if it is not used

TheLegendaryJedi, 2021-03-25:
Is there a way to debug it? Because it does not give me anything on the console.

bresch, 2021-03-25:
You could add some printfs indicators in the code to know where it stops.

TheLegendaryJedi, 2021-04-01:
Hi @bresch, I made some prints like you suggested and figured that in `control.cpp` the `_flow_sample_delayed` never has good samples.

Basically the quality is always 0, and the dt is always 0 too. Any suggestion on what might be the problem?

`INFO  [ecl/EKF] ------------------SAMPLE-------------------`
`INFO  [ecl/EKF] is_quality_good: 0`
`INFO  [ecl/EKF]   _flow_sample_delayed.quality >= _params.flow_qual_min`
`INFO  [ecl/EKF]     _flow_sample_delayed.quality: 0`
`INFO  [ecl/EKF]     _params.flow_qual_min: 1`
`INFO  [ecl/EKF] is_magnitude_good: 1`
`INFO  [ecl/EKF] is_tilt_good: 1`
`INFO  [ecl/EKF] is_delta_time_good: 0`
`INFO  [ecl/EKF]   _flow_sample_delayed.dt >= delta_time_min && _flow_sample_delayed.dt <= delta_time_max`
`INFO  [ecl/EKF]     _flow_sample_delayed.dt: 0.000000`
`INFO  [ecl/EKF]     delta_time_min: 0.051943`
`INFO  [ecl/EKF]     _flow_sample_delayed.dt: 0.000000`
`INFO  [ecl/EKF]     delta_time_max: 0.077914`
`INFO  [ecl/EKF] is_body_rate_comp_available: 1`
`ERROR [ekf2] 0 - distance_sensor lost, generation 385 -> 388`

TheLegendaryJedi, 2021-04-07:
https://review.px4.io/plot_app?log=7ab4c688-20db-4a0f-a5d0-bf9d7d8b229d

bresch, 2021-04-07:
Thanks for the log, I'll have a look. I also received a CrazyFlie 2.1 yesterday so I should be able to reproduce your issue soon.

TheLegendaryJedi, 2021-04-07:
> Thanks for the log, I'll have a look. I also received a CrazyFlie 2.1 yesterday so I should be able to reproduce your issue soon.

Nice! It's preatty awesome.

Just one update. During dev call Matthias and Dagar told I needed to enable pmw3901 parameter and I did. Right now I've manually started the pmw3901:
`pmw3901 start -s -b 1`

`INFO  [SPI_I2C] Running on SPI Bus 1`
`pmw3901: read: 3077 events, 740455us elapsed, 240.64us avg, min 87us max 16947us 957.049us rms`
`pmw3901: com err: 0 events`

But I still can't change to position mode or hold mode.

We have an altitude estimation.

https://logs.px4.io/plot_app?log=948a52f3-b417-4a25-a674-49be2c0451d2

bresch, 2021-04-30:
Current status: I'm able to get a meaningful local position estimate when doing handheld tests (in the picture below, I'm moving the drone roughly in a square shape and I finish by a circle).

I'll make a flight test when possible

[image]
log: https://logs.px4.io/plot_app?log=78b1b6bd-3527-4ca1-bb57-b81a490e7aff

TheLegendaryJedi, 2021-04-30:
Nice! Thanks!

Can you send me your parameter file?
I want to test it too.

bresch, 2021-05-01:
@TheLegendaryJedi Basically I had to:
- remove ` && _is_yaw_fusion_inhibited` from https://github.com/PX4/PX4-ECL/blob/a7b8afe420f438554ad90bcba0f1f4872325e75b/EKF/ekf_helper.cpp#L518
- start the pmw3901 with the `-s` flag (as you mentioned before)
- set `EKF2_MAG_TYPE` to mc custom
- set `SENS_FLOW_ROT` to 180 degrees

TheLegendaryJedi, 2021-05-01:
Thanks a lot! I'll do it. Probably only Tuesday next week.

TheLegendaryJedi, 2021-05-06:
> @TheLegendaryJedi Basically I had to:
> 
> * remove ` && _is_yaw_fusion_inhibited` from https://github.com/PX4/PX4-ECL/blob/a7b8afe420f438554ad90bcba0f1f4872325e75b/EKF/ekf_helper.cpp#L518
> * start the pmw3901 with the `-s` flag (as you mentioned before)
> * set `EKF2_MAG_TYPE` to mc custom
> * set `SENS_FLOW_ROT` to 180 degrees
> 
> Unfortunately it seems that the LiPo died so I cannot do a flight test

Hi @bresch, just tested it with 1.12 beta 4 and your recommendations.

The position estimate is very slow. Everything is happening with a delay, for example, when you are in position mode all your joystick commands happen with a delay (roll, pitch, yaw, throttle). If I try to auto takeoff and hold (0.5m) it goes up like 4 meters and won't stop, I have to kill switch it. Could it be that the EKF2 with position estimate is too much for STM32F405?

My [flight log:](https://logs.px4.io/plot_app?log=1e8c1745-f479-4f28-b0b8-6372f70829ab)

bresch, 2021-05-11:
In your log the drone is in position mode and the throttle stick is at 58% which is just enough to takeoff and then the drone lands again due to the error in velocity estimate. I don't see the auto takeoff.
For the rest, the baro is highly affected and you could switch to rng aid only (`HGT_MODE`) since this is an indoor application. Also, `EKF2_MIN_RNG` is set to 10cm, which is a lot for this drone. You can set it to almost 0

[image]

TheLegendaryJedi, 2021-05-13:
Just tried your recomendations and takeoff still drifts a lot:
https://review.px4.io/plot_app?log=09c156ed-f204-4470-91f6-278a01a7c96b

Takeoff with crash:
https://review.px4.io/plot_app?log=8ad23f59-372f-4d0b-9449-f621e4be6667

Regarding position mode it's the same, a lot of lag from sending the commands and the actual response from the drone:
https://review.px4.io/plot_app?log=13250d4a-4f14-414d-ae43-2c1531e3bf79

Thanks!

slgrobotics, 2024-05-16:
@bresch @TheLegendaryJedi  - if you still have CrazyFlie 2.1 hardware, I would appreciate testing the BMI088/I2C IMU (not necessarily in flight, just move it by hand). I believe the problems you had were in part due to its malfunction.

That's for PR https://github.com/PX4/PX4-Autopilot/pull/23141

bresch, 2024-06-13:
@slgrobotics Unfortunately my CrazyFlie doesn't connect to QGC anymore for some reason...

slgrobotics, 2024-06-14:
@bresch  - well, the BMI088/I2C thing seems marginal anyway, and maybe CrazyFlie shouldn't be officially supported (not my call of course). But my last comment here https://github.com/PX4/PX4-Autopilot/pull/23141 about SPI-connected devices may deserve consideration (```#4```). It is worth handling those FFFF's properly and re-testing SPI devices (see https://github.com/PX4/PX4-Autopilot/issues/23111 for full description of the problem).

slgrobotics, 2024-12-04:
@bresch @dagar  - my last comment above, that's the common bug between I2C and SPI, maybe deserving a second look.