PX4/PX4-Autopilot #17087: FMUK66 revE log files format issue

dk7xe, 2021-03-11:
**Describe the bug**
log files of FMUK66 revE seems to have an format issue and can not be loaded on to logs.px4.io for analysis.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable logging on FMUK66 revE (nxp_fmuk66-e_default target)
2. Power drone, fly, .. land, power off.
3. Download log file via USB
4. Try to upload log file to https://logs.px4.io

**Expected behavior**
Log files accepted by https://logs.px4.io

**Log Files and Screenshots**
Log file examples
[ID10_TestDrone3.zip](https://github.com/PX4/PX4-Autopilot/files/6120873/ID10_TestDrone3.zip)

Screenshot
[image]

dk7xe, 2021-03-11:
@davids5 may you please look into that

davids5, 2021-03-11:
I can not get to it at the moment, A quick fix may be enabling CONFIG_MMCSD_MULTIBLOCK_DISABLE

closed 2021-03-12 by LorenzMeier