PX4/PX4-Autopilot #17091: multiple vehicle SITL simulation does not work with iris_rtps model

96moustafa, 2021-03-11:
**Describe the bug**
I am trying to run multi-vehicle SITL simulation as documented [here](https://docs.px4.io/master/en/simulation/multi_vehicle_simulation_gazebo.html#build-and-test-rtps-dds). The simulation window in Gazebo opens, but no vehicles exits and QGC cannot detect any.

**To Reproduce**
follow the RTPS documentation [here](https://docs.px4.io/master/en/simulation/multi_vehicle_simulation_gazebo.html#build-and-test-rtps-dds).

**Expected behavior**
Simulation starts with the correct number of vehicles and be able to connect to them using QGC.

**Additional context**
This is the terminal output after running `./Tools/gazebo_sitl_multiple_run.sh -m iris_rtps -t px4_sitl_rtps -n 2`

<pre>killing running instances
GAZEBO_PLUGIN_PATH :/home/mamado/latest_src/PX4-Autopilot/Tools/../build/px4_sitl_rtps/build_gazebo
GAZEBO_MODEL_PATH :/home/mamado/latest_src/PX4-Autopilot/Tools/../Tools/sitl_gazebo/models
LD_LIBRARY_PATH /opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib:/home/mamado/latest_src/PX4-Autopilot/Tools/../build/px4_sitl_rtps/build_gazebo
Starting gazebo
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

<font color="#4E9A06"><b>[Msg] Waiting for master.</b></font>
<font color="#4E9A06"><b>[Msg] Connected to gazebo master @ http://127.0.0.1:11345</b></font>
<font color="#4E9A06"><b>[Msg] Publicized address: 192.168.158.87</b></font>
<font color="#4E9A06"><b>[Msg] Loading world file [/home/mamado/latest_src/PX4-Autopilot/Tools/../Tools/sitl_gazebo/worlds/empty.world]</b></font>
<font color="#4E9A06"><b>starting instance 0 in /home/mamado/latest_src/PX4-Autopilot/build/px4_sitl_rtps/instance_0</b></font>
<font color="#4E9A06"><b>Traceback (most recent call last):</b></font>
<font color="#4E9A06"><b>  File &quot;/home/mamado/latest_src/PX4-Autopilot/Tools/../Tools/sitl_gazebo/scripts/jinja_gen.py&quot;, line 38, in &lt;module&gt;</b></font>
<font color="#4E9A06"><b>    template = env.get_template(os.path.relpath(args.filename, args.env_dir))</b></font>
<font color="#4E9A06"><b>  File &quot;/home/mamado/.local/lib/python3.8/site-packages/jinja2/environment.py&quot;, line 883, in get_template</b></font>
<font color="#4E9A06"><b>    return self._load_template(name, self.make_globals(globals))</b></font>
<font color="#4E9A06"><b>  File &quot;/home/mamado/.local/lib/python3.8/site-packages/jinja2/environment.py&quot;, line 857, in _load_template</b></font>
<font color="#4E9A06"><b>    template = self.loader.load(self, name, globals)</b></font>
<font color="#4E9A06"><b>  File &quot;/home/mamado/.local/lib/python3.8/site-packages/jinja2/loaders.py&quot;, line 115, in load</b></font>
<font color="#4E9A06"><b>    source, filename, uptodate = self.get_source(environment, name)</b></font>
<font color="#4E9A06"><b>  File &quot;/home/mamado/.local/lib/python3.8/site-packages/jinja2/loaders.py&quot;, line 197, in get_source</b></font>
<font color="#4E9A06"><b>    raise TemplateNotFound(template)</b></font>
<font color="#4E9A06"><b>jinja2.exceptions.TemplateNotFound: models/iris_rtps/iris_rtps.sdf.jinja</b></font>
<font color="#4E9A06"><b>Spawning iris_rtps_0 at 0.0 0</b></font>
<font color="#4E9A06"><b>Error: Unable to open file[/tmp/iris_rtps_0.sdf]</b></font>
<font color="#4E9A06"><b>starting instance 1 in /home/mamado/latest_src/PX4-Autopilot/build/px4_sitl_rtps/instance_1</b></font>
<font color="#4E9A06"><b>Traceback (most recent call last):</b></font>
<font color="#4E9A06"><b>  File &quot;/home/mamado/latest_src/PX4-Autopilot/Tools/../Tools/sitl_gazebo/scripts/jinja_gen.py&quot;, line 38, in &lt;module&gt;</b></font>
<font color="#4E9A06"><b>    template = env.get_template(os.path.relpath(args.filename, args.env_dir))</b></font>
<font color="#4E9A06"><b>  File &quot;/home/mamado/.local/lib/python3.8/site-packages/jinja2/environment.py&quot;, line 883, in get_template</b></font>
<font color="#4E9A06"><b>    return self._load_template(name, self.make_globals(globals))</b></font>
<font color="#4E9A06"><b>  File &quot;/home/mamado/.local/lib/python3.8/site-packages/jinja2/environment.py&quot;, line 857, in _load_template</b></font>
<font color="#4E9A06"><b>    template = self.loader.load(self, name, globals)</b></font>
<font color="#4E9A06"><b>  File &quot;/home/mamado/.local/lib/python3.8/site-packages/jinja2/loaders.py&quot;, line 115, in load</b></font>
<font color="#4E9A06"><b>    source, filename, uptodate = self.get_source(environment, name)</b></font>
<font color="#4E9A06"><b>  File &quot;/home/mamado/.local/lib/python3.8/site-packages/jinja2/loaders.py&quot;, line 197, in get_source</b></font>
<font color="#4E9A06"><b>    raise TemplateNotFound(template)</b></font>
<font color="#4E9A06"><b>jinja2.exceptions.TemplateNotFound: models/iris_rtps/iris_rtps.sdf.jinja</b></font>
<font color="#4E9A06"><b>Spawning iris_rtps_1 at 0.0 3</b></font>
<font color="#4E9A06"><b>Error: Unable to open file[/tmp/iris_rtps_1.sdf]</b></font>
<font color="#4E9A06"><b>Starting gazebo client</b></font>

</pre>

Jaeyoung-Lim, 2021-03-11:
@stmoon Do you have any insights?

stmoon, 2021-03-12:
Hi,

Actually, I updated the method for the multiple vehicle like below.

FROM
```
./Tools/gabo_sitl_multiple_run.sh  -m iris_rtps -t px4_sitl_rtps -n 4
```
TO.
```
Tools/gazebo_sitl_multiple_run.sh -t px4_sitl_rtps -m iris -l rtps -n 4
```

I will update the documentation ASAP.

I apologize for the inconvenience.

closed 2021-03-14 by Jaeyoung-Lim