PX4/PX4-Autopilot #17093: px4io failed start on Cube Orange

ArkadiuszNiemiec, 2021-03-11:
**Describe the bug**
Sometimes there is no RC connection with Cube Orange, this is the output of the px4io status:
```
nsh> px4io status
ERROR [px4io] not started
nsh> px4io start
ERROR [px4io] Failed to communicate with IO, abort.
ERROR [px4io] driver init failed
```

Rebooting the Cube does not help, to fix a problem we have to power cycle the UAV.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the RC
2. Power on the Cube
3. Realize that there is no control and no RC failsafe
4. Check mavlink console for px4io status

**Expected behavior**
I expect the IO board to always work

**Additional context**
PX4 v1.11.0dev (we have made a one-line change to ecl package).

ArkadiuszNiemiec, 2021-03-11:
After another reboot the px4io was working but no value was changing, after `px4io stop` and `px4io start` we got another:
```
nsh> px4io start
ERROR [px4io] Failed to communicate with IO, abort.
ERROR [px4io] driver init failed
```

LorenzMeier, 2021-03-11:
@ArkadiuszNiemiec Could you please install master and re-test? I've also pinged you on Slack.

ArkadiuszNiemiec, 2021-03-12:
Looks like master helped with the issue, we will test it further next week and close the issue if there will be no problem.

ArkadiuszNiemiec, 2021-03-18:
After tens of flights and restarts I can assume that the issue is fixed. Thank you.

closed 2021-03-18 by ArkadiuszNiemiec