PX4/PX4-Autopilot #17097: Px4 log unknown date in QGC

aet2230, 2021-03-11:
Recently I started to use Cube Orange and I get logs with unknown date in QGC.
When I look at SD card, logs do have correct date (foldername) and time (filename).
Usually such issue happens when I erase logs and perform flights til firs connection with QGC.
Once QGC is connected once, further logs are displayed.
Attached is the screenshot that describes the issue: I entered nsh an you may see that in logdata.txt, the logname is ok, but its timestamp represents Saturday, January 1, 2000

[image]

julianoes, 2023-04-03:
I tried to reproduce this but failed. For me it seems correct. Do you have any more hints what to try? Does it still happen?

aet2230, 2023-04-03:
Too old, may be closed

closed 2023-04-03 by julianoes

vacoff, 2024-08-14:
I have come across the same issue today, new cube orange had the exact same issue. it's the second flight of the platform