PX4/PX4-Autopilot #17114: Mavros does not work in multiple vehicle simulation with different models

cserko, 2021-03-14:
**Describe the bug**
A clear and concise description of the bug.
I tried to use multi_uav_mavros_sitl.launch with the iris_rplidar model. However, when I try to echo the uav topics e.g. /uav1/mavros/local_position/pose, only the topics which have the namespace uav0 successfully return the callback values.

**Expected behavior**
A clear and concise description of what you expected to happen.
All the mavros topics with the specified group namespaces should have responded to rostopic echo commands for example, 
[image]

**Log Files and Screenshots**
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/carousel/Documents/Firmware/build/px4_sitl_default/tmp && /home/carousel/Documents/Firmware/Tools...none gazebo none none /home/carousel/Documents/Firmware /home/carousel/Documents/Firmware/build/px4_sitl_default
Not running simulation (DONT_RUN is set).
GAZEBO_PLUGIN_PATH :/home/carousel/Documents/Firmware/build/px4_sitl_default/build_gazebo:/home/carousel/Documents/Firmware/build/px4_sitl_default/build_gazebo:/home/carousel/Documents/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/carousel/avoidance/catkin_ws/src/avoidance/avoidance/sim/models:/home/carousel/avoidance/catkin_ws/src/avoidance/avoidance/sim/worlds:/home/carousel/Documents/Firmware/Tools/sitl_gazebo/models:/home/carousel/Documents/Firmware/Tools/sitl_gazebo/models:/home/carousel/Documents/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/carousel/avoidance/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/carousel/Documents/Firmware/build/px4_sitl_default/build_gazebo:/home/carousel/Documents/Firmware/build/px4_sitl_default/build_gazebo:/home/carousel/Documents/Firmware/build/px4_sitl_default/build_gazebo
... logging to /home/carousel/.ros/log/bf451b5e-8505-11eb-b0ed-68ecc5c6d8d8/roslaunch-carousel-1746.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://carousel:39237/

SUMMARY
========

CLEAR PARAMETERS
 * /uav1/mavros/
 * /uav2/mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /uav1/mavros/cmd/use_comp_id_system_control: False
 * /uav1/mavros/conn/heartbeat_rate: 1.0
 * /uav1/mavros/conn/system_time_rate: 1.0
 * /uav1/mavros/conn/timeout: 10.0
 * /uav1/mavros/conn/timesync_rate: 10.0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /uav1/mavros/distance_sensor/laser_1_sub/id: 3
 * /uav1/mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /uav1/mavros/distance_sensor/laser_1_sub/subscriber: True
 * /uav1/mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /uav1/mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /uav1/mavros/distance_sensor/lidarlite_pub/id: 1
 * /uav1/mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /uav1/mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /uav1/mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /uav1/mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /uav1/mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /uav1/mavros/distance_sensor/sonar_1_sub/id: 2
 * /uav1/mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /uav1/mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /uav1/mavros/fake_gps/eph: 2.0
 * /uav1/mavros/fake_gps/epv: 2.0
 * /uav1/mavros/fake_gps/fix_type: 3
 * /uav1/mavros/fake_gps/geo_origin/alt: 408.0
 * /uav1/mavros/fake_gps/geo_origin/lat: 47.3667
 * /uav1/mavros/fake_gps/geo_origin/lon: 8.55
 * /uav1/mavros/fake_gps/gps_rate: 5.0
 * /uav1/mavros/fake_gps/mocap_transform: True
[... 657 lines cut ...]
INFO  [ecl/EKF] 7092000: reset velocity to GPS
INFO  [ecl/EKF] 7092000: starting GPS fusion
INFO  [ecl/EKF] 7092000: reset position to GPS
INFO  [ecl/EKF] 7092000: reset velocity to GPS
INFO  [ecl/EKF] 7092000: starting GPS fusion
INFO  [ecl/EKF] 7096000: reset position to GPS
INFO  [ecl/EKF] 7096000: reset velocity to GPS
INFO  [ecl/EKF] 7096000: starting GPS fusion
INFO  [ecl/EKF] 7096000: reset position to GPS
INFO  [ecl/EKF] 7096000: reset velocity to GPS
INFO  [ecl/EKF] 7096000: starting GPS fusion
INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative
[ INFO] [1615754549.009995473, 16.328000000]: WP: mission received

*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/master/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.
[image]

Jaeyoung-Lim, 2021-03-14:
@cserko Have you checked if it works with the default `iris` model?

cserko, 2021-03-14:
Yes, it works with the regular model. Actually, I've recently found out if I add the rplidar plugin as whole to the iris.sdf.jinja, and start the simulation using the regular iris model solved the topic issue. However, as I said, I couldn't figure out how to start like rplidar/rplidar.sdf, and access the mavros topics. It works now but it would be very inconvenient to change one file instead of using include tags.

For future reference, I am now having another issue with the rplidar model. Even though I mimicked the rplidar.sdf, and iris_rplidar.sdf files while adding to the iris.sdf.jinja, the model does not stay stable. Here is the updated iris.sdf.jinja 
[iris.sdf.jinja.txt](https://github.com/PX4/PX4-Autopilot/files/6137837/iris.sdf.jinja.txt)

Thanks for the reply. Best.

closed 2021-04-22 by Jaeyoung-Lim