PX4/PX4-Autopilot #17119: Jitter in mode of Altitude at MC(VTOL)

Kim-zhi-jiang, 2021-03-15:
Hello there

I have a problem in version 1.11.3, I will test again in version 1.12 tomorrow.
MC(VTOL) in the mode of Altitude, only the forward direction jitters, and strangely it is no problem to fly backwards to the left and to the right.

[image]

jinger26, 2021-03-15:
@sfuhrer @RomanBapst are you able to take a look here?

Jaeyoung-Lim, 2021-03-15:
@Kim-zhi-jiang Can you share the full log?

Kim-zhi-jiang, 2021-03-15:
> @Kim-zhi-jiang Can you share the full log?

https://review.px4.io/plot_app?log=3d4a955f-41ed-4f71-93b9-1aa45727bcab

Just uploaded.

Kim-zhi-jiang, 2021-03-15:
I tried to find the source of the problem, but unfortunately.
Forward  direction have airspeed...

sfuhrer, 2021-03-16:
Looks to me like it struggles a lot when flying (hovering) with high roll/pitch angles.
@Kim-zhi-jiang can you share a flight done in Stabilized and Position mode? That would help finding out if the issue is indeed coming from the altitude controller, and/or if related to high attitude angles.

Kim-zhi-jiang, 2021-03-17:
@sfuhrer 

MC_ROLLRATE_MAX 120-150
MC_PITCHRATE_MAX 120-150

Forward,Backward,Left,Right it's OK in Stabilized and Position mode
Jitters only forward in Altitude mode

I think the plane will crashed with this problem in windy weather.

Stabilized Position and Altitude log:
https://review.px4.io/plot_app?log=3cacd01b-213f-4092-9f12-8330b416296f

Kim-zhi-jiang, 2021-03-17:
@all
The controller was wrong if Position change to Stabilized mode.
Also move the plane  in Stabilized mode.

crying

Kim-zhi-jiang, 2021-03-17:
> Looks to me like it struggles a lot when flying (hovering) with high roll/pitch angles.
> @Kim-zhi-jiang can you share a flight done in Stabilized and Position mode? That would help finding out if the issue is indeed coming from the altitude controller, and/or if related to high attitude angles.
> Does the vehicle have a large wing area?

If forward direction speed more than 11m/s then also jitters in Position mode.

closed 2021-03-18 by Kim-zhi-jiang