PX4/PX4-Autopilot #17120: CUAV smart battery 🔋 implementation

jinger26, 2021-03-15:
CUAV is implementing a smart battery, and looking to implement the following messages

Battery status name | CUAV usage| Included in the UAVCAN spec & standard| Included in the smart battery standard
------------ | ------------- | ------------- | -------------
timestamp | no | - | -
voltage_v | yes | yes | yes
voltage_filtered_v| yes | yes | yes
current_a| yes | yes | yes
current_filtered_a| yes | yes | yes
average_current_a| yes | no | yes
discharged_mah| yes| no | yes
remaining| yes| yes | yes
scale| no | - | -
temperature| yes| yes | yes
cell_count| yes| no | yes
voltage_cell_v[10]| yes| no | yes
max_cell_voltage_delta| yes| no | yes
capacity| yes| yes | yes
cycle_count| yes| no | yes
run_time_to_empty| yes| no | yes
average_time_to_empty| no | - | -
serial_number| yes| yes | yes
manufacture_date| yes| no | yes
state_of_health| yes| yes | yes
max_error| yes| yes | yes
interface_error| no | - | -
connected| no | - | -
source| no | - | -
priority| no | - | -
id| no | - | -
is_powering_off| yes| no | yes
warning| no | - | -
_padding0[4]| no | - | -

This is the tracker to add the missing messages that is needed for productization:

- [ ] average_current_a
- [ ] discharged_mah
- [ ] cell_count
- [ ] voltage_cell_v[10]
- [ ] max_cell_voltage_delta
- [ ] cycle_count
- [ ] run_time_to_empty
- [ ] manufacture_date
- [ ] is_powering_off

jinger26, 2021-03-15:
@Igor-Misic @LorenzMeier  if we can make one reference example, then we should be able to enable CUAV team to do the rest.

jinger26, 2021-03-15:
https://github.com/PX4/PX4-Autopilot/pull/17104

mxiaogit, 2021-03-18:
We want to realize the information in the battery status:

PX4 battery status information | CUAV usage | Included in the UAVCAN spec & standard | Description
-- | -- | -- | --
timestamp | yes | no | time since system start (microseconds)
voltage_v | yes | yes | Battery voltage in volts, 0 if unknown
voltage_filtered_v | yes | yes | Battery voltage in volts, filtered, 0 if unknown
current_a | yes | yes | Battery current in amperes, -1 if unknown
current_filtered_a | yes | yes | Battery current in amperes, filtered, 0 if unknown
average_current_a | yes | no | Battery current average in amperes, -1 if unknown
discharged_mah | yes | no | Discharged amount in mAh, -1 if unknown
remaining | yes | yes | From 1 to 0, -1 if unknown
scale | no | no | Power scaling factor, >= 1, or -1 if unknown
temperature | yes | yes | temperature of the battery. NaN if unknown
cell_count | yes | no | Number of cells
connected | yes | no | Whether or not a battery is connected, based on a voltage threshold
source | yes | no | Battery source
priority | no | no | Zero based priority is the connection on the Power Controller V1..Vn AKA BrickN-1
capacity | yes | yes | actual capacity of the battery
cycle_count | yes | no | number of discharge cycles the battery has experienced
run_time_to_empty | yes | no | predicted remaining battery capacity based on the present rate of discharge in min
average_time_to_empty | yes | no | predicted remaining battery capacity based on the present rate of discharge in min
serial_number | yes | yes | serial number of the battery pack
manufacture_date | yes | no | manufacture date, part of serial number of the battery pack. formated as: Day + Month×32 + (Year–1980)×512
state_of_health | yes | yes | state of health. FullChargeCapacity/DesignCapacity.
max_error | yes | yes | max error, expected margin of error in % in the state-of-charge calculation with a range of 1 to 100%
id | yes | no | ID number of a battery. Should be unique and consistent for the lifetime of a vehicle. 1-indexed.
interface_error | yes | no | SMBUS interface error counter
voltage_cell_v[10] | yes | no | Battery individual cell voltages
max_cell_voltage_delta | yes | no | Max difference between individual cell voltages
is_powering_off | yes | no | Power off event imminent indication, false if unknown
warning | yes | no | current battery warning

Among them, the number of battery cells in voltage_cell_v [10] is insufficient. In mavlink 2 it supports up to 14 batteries, and we hope to increase it from 10 to 14, which is voltage_cell_v [14].

We also hope to add the following information in battery status:

Added information | Description
-- | --
average_power | The average power of the current discharge
available_energy | The predicted charge or energy remaining in the battery
remaining_capacity | The compensated battery capacity remaining
full_charge_capacity | The compensated capacity of the battery when fully charged
design_capacity | The design capacity of the battery
average_time_to_full | The predicted remaining time until the battery reaches full charge
over_discharge_count | Number of battery overdischarge
nominal_voltage | Nominal voltage of the battery pack

So far, we have initially completed the required information.
The link is here: https://github.com/PX4/PX4-Autopilot/pull/17104

mxiaogit, 2021-03-18:
Test results on 3/17 by @mxiaogit  :

**Test environment :（测试环境）

Firmware version :（固件版本）
59f361419c9f68fbb243b8b4477548b631b3bd24

Test hardware :（测试硬件）
CUAV V5
CUAV CBAT

Firmware modification :（固件修改）
/Firmware/src/drivers/uavcan/dsdl/cuav/equipment/power/20300.CBAT.uavcan
Add UAVCAN CBAT protocol file
[image]

/Firmware/src/drivers/uavcan/CMakeLists.txt
Add the path of the 20300.CBAT.uavcan and cbat.c files
[image]
[image]

/Firmware/msg/battery_status.msg
Add extended information for backup
[image]

/Firmware/src/drivers/uavcan/sensors/cbat.cpp
Add CUAV smart battery support
[image]
[image]
[image]
[image]

/Firmware/src/drivers/uavcan/sensors/cbat.hpp
Add CUAV smart battery support
[image]

/Firmware/src/drivers/uavcan/uavcan_params.c
Add UAVCAN_BAT_MON parameter to switch to UAVCAN CBAT protocol, the default is UAVCAN BatteryInfo protocol.
[image]

/Firmware/src/drivers/uavcan/sensors/sensor_bridge.cpp
Add UAVCAN_BAT_MON parameter to switch to UAVCAN CBAT protocol, the default is UAVCAN BatteryInfo protocol.
[image]

**Test process :（测试流程）

Use QGC to set the parameter UAVCAN_BAT_MON=1, use UAVCAN CBAT protocol （使用QGC设置参数UAVCAN_BAT_MON=1，使用UAVCAN CBAT协议）
[image]

Use QGC to view battery status information （使用QGC查看电池状态信息）
[image]
[image]

**Result :（结果）
In this firmware version(59f361419c9f68fbb243b8b4477548b631b3bd24), the display of battery cell voltage and remaining use time information is provided(在这个固件版本中(59f361419c9f68fbb243b8b4477548b631b3bd24)，提供了电池单片电压和剩余使用时间信息的显示)

mxiaogit, 2021-03-18:
The driver has been initially completed. @LorenzMeier @dagar

mxiaogit, 2021-04-12:
The driver has been completed.

closed 2025-12-07 by github-actions[bot]