PX4/PX4-Autopilot #17122: FW trajectory tracking using OFFBOARD mode

AndreRigi, 2021-03-15:
Hello, 

We are using an external module to handle our missions. It publishes mavros_msgs::GlobalPositionTarget which are used by PX4 using the offboard mode. Since the merge #16739  (@dagar, @Jaeyoung-Lim), the drone is no longer tracking correctly the trajectory setpoints when in FW. We are using a VTOL drone and the initial part of the flight, FW transition included, is working properly. Once in FW,  the drone is however no longer tracking correctly the trajectory setpoints. 
We also noticed that during the transition to FW, the trajectory setpoints are set to NaN for a short period. 

Steps to reproduce the behavior:
1. VTOL drone, in offboard_control_mode/position.
1. Uploaded offboard mission here attached. 
3. Drone is flying towards the front transition end waypoint and has already switched properly to FW. Once in FW, it stops tracking the trajectory setpoints.

[image]

Log: https://logs.px4.io/plot_app?log=0ab8684d-5323-4032-b6e5-a807dd1fc564
Mission plan: [mission.zip](https://github.com/PX4/PX4-Autopilot/files/6140868/mission.zip)

Jaeyoung-Lim, 2021-03-15:
@AndreRigi I guess you mean that you are using the ` SET_POSITION_TARGET_GLOBAL_INT` mavlink message?

what are you setting as `MAV_FRAME` ?

Jaeyoung-Lim, 2021-03-15:
I can reproduce this problem and this is because the current poistion setpoint type is currently set as `SETPOINT_TYPE_IDLE`

@dagar As far as I understand, I think the intention was to remove the setpoint types from the position setpoints in https://github.com/PX4/PX4-Autopilot/pull/16739 (at least it has been removed from the mavlink receiver.

Do you have any insights on where this should be fixed? - should we define the setpoint type somewhere along the pipeline? or should we change the fixedwing position controller?

closed 2021-03-16 by Jaeyoung-Lim